stm32-rs/stm32f4xx-hal #1: Documentation

timbod7, 2018-10-04:
The online documentation at (https://docs.rs/stm32f4xx-hal/0.1.0/stm32f4xx_hal), linked from (https://crates.io/crates/stm32f4xx-hal), does not exist.

Is this much effort to fix? Is this something I can assist with?

therealprof, 2018-10-05:
I have a suspicion it might fail for the same reason why other crates couldn't get an updated version of the documentation rendered, cf. https://docs.rs/crate/stm32f042-hal/0.5.1/builds/118064

I think I've seen a ticket somewhere about this but at the moment I'm a bit clueless what we can do to fix it.

jonas-schievink, 2018-10-07:
Seems to me that the Rust toolchain used by docs.rs needs to be updated. cc https://github.com/onur/docs.rs/issues/233

therealprof, 2018-11-03:
@timbod7 Works now, so I'm closing this.

closed 2018-11-03 by therealprof

WGH-, 2019-01-01:
https://docs.rs/stm32f4xx-hal/0.2.6/stm32f4xx_hal/ has a link to https://docs.rs/stm32f4/0.4.0/stm32f4/index.html, which is broken right now. Is it the same problem?

therealprof, 2019-01-02:
@WGH- stm32f4 does not have any docs at all. Not sure how to get that properly cleaned up.

adamgreig, 2019-01-02:
The build logs are [here](https://docs.rs/crate/stm32f4/0.4.0/builds/126965), looks like the doc process was oom killed. We could try having fewer devices in the docs features to try and reduce the memory load, but it's not a great long term solution. See also [stm32-rs#3](https://github.com/stm32-rs/stm32-rs/issues/3)