stm32-rs/stm32f4xx-hal #3: control of USART serial format

timbod7, 2018-10-07:
I've started attempting to use the serial abstraction, and immediately needed to tweak the flags. The existing code:

```
        usart
            .cr1
            .write(|w| w.ue().set_bit().te().set_bit().re().set_bit());
```

I needed to change to:

```
        usart
            .cr1
            .write(|w| w
              .ue().set_bit()
              .te().set_bit()
              .re().set_bit()
              .m().set_bit()    // 9 data bits
              .pce().set_bit()  // enable parity
              .ps().clear_bit() // even parity
              );

        usart.cr2.write( |w| w
          .stop().bits(2)    // 2 stops bits
        );
```

It didn't seem possible to set these flags prior or after the call to `Serial::usart2()`.  Have you any ideas as to how you would generalise the API to allow control over the serialized format? Perhaps an extra parameter with options to control parity, number of stop bits etc?

gbip, 2018-10-07:
I think the builder pattern is most rustic way of doing this kind of thing :smile:

therealprof, 2018-10-10:
@timbod7 Please close this if you're happy with the current state.

NB: You can also add a "Fixes #<number>" to your PR and then the issue will autoclose when the PR is merged.

closed 2018-10-10 by timbod7