stm32-rs/stm32f4xx-hal #4: Is this a good place for the stm32f446 hal ?

gbip, 2018-10-07:
Hi there !
I have a partial hal for the stm32f446 forked from the one from [dkarwowski](https://github.com/dkarwowski/stm32f429x) but I am missing the time to implement all the features...
Could I reuse your implementation and add support for the stm32f446 in this repo ?
Thanks !

therealprof, 2018-10-07:
@gbip Absolutely! PRs very welcome. I also noticed @astro has done some work for the STM32F429 in `stm32f429-hal` which I wanted to investigate and take over. I've started this with a minimal but tested to work feature set and wanted to extend gradually to support more peripherals and devices.

birkenfeld, 2018-10-31:
@therealprof I've made my own changes on `stm32f429` and was wondering which crate to base my future work on. I take it this one makes more sense?

therealprof, 2018-10-31:
@birkenfeld Well, that was the secret plan to make it most ubiquitous. ;)

aurelj, 2019-01-26:
As this crate is now supporting the stm32f446, I think this clearly answer the initial question and this issue can be closed.

closed 2019-01-26 by aurelj