stm32-rs/stm32f4xx-hal #5: Examples

petoknm, 2018-10-07:
I am trying to use serial but failing to do so. [STM32F4Discovery board]
```
#[entry]
fn main() -> ! {
    let p = hal::stm32::Peripherals::take().unwrap();

    let rcc = p.RCC.constrain();
    let clocks = rcc.cfgr
        .hclk(8.mhz())
        .sysclk(168.mhz())
        .freeze();
    let gpioa = p.GPIOA.split();

    // USART1
    let tx = gpioa.pa9.into_alternate_af7();
    let rx = gpioa.pa10.into_alternate_af7();

    let s = Serial::usart1(p.USART1, (tx, rx), 9600.bps(), clocks);
    let (mut tx, _) = s.split();

    loop {
        block!(tx.write(b'x'));
    }
}
```

And all I get on the serial is unreadable mess (0xF8, 0xF8, 0xF8, ...).
I think it would be nice to have examples that were tested to work.

petoknm, 2018-10-07:
Most likely my fault. Switched to using PB6/7 and it works. Maybe the board has something attached to PA9/10 and I'm too lazy to check the datasheet. 
Great job and thanks for the library :+1:

closed 2018-10-07 by petoknm

therealprof, 2018-10-07:
@petoknm Thanks for using this. Just as a side note, I've also created the `stm32f407g-disc` and `stm32f429i-disc` board support crates using this crate and containing examples which are tested to work on those boards.

chocol4te, 2019-01-21:
Just in case anyone wants another example, I've got RTFM, external interrupts and PWM in/out working on my STM32F411E Discovery [here](https://github.com/chocol4te/crankshaft/blob/2609023086b4a52f6489ec65a2fada7410480716/src/main.rs).