stm32-rs/stm32f4xx-hal #12: error: symbol `DEVICE_PERIPHERALS` is already defined

jajuul, 2018-10-10:
I cannot seem to compile the crate when using the stm32f429 feature. I have tried to use your crate in a project but could not compile it. So I tried cloning your repo to see if I could compile that. If I try to compile using this command:

`cargo build --features stm32f407,rt`

it works. If I try with the stm32f429 like this:

`cargo build --features stm32f429,rt`

it fails giving the error below:

```
error: symbol `DEVICE_PERIPHERALS` is already defined
    --> C:\Users\Ju\.cargo\registry\src\github.com-1ecc6299db9ec823\stm32f4-0.3.2\src\stm32f429\mod.rs:2206:1
     |
2206 | static mut DEVICE_PERIPHERALS: bool = false;
     | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

error: aborting due to previous error
```

I'm new to embedded rust and rust in general, so sorry if I am missing something obvious.

therealprof, 2018-10-10:
@jajuul You need to disable the `default-features` with `--no-default-features`.

jajuul, 2018-10-11:
That worked.
Thank you for the fast reply.

closed 2018-10-11 by jajuul