stm32-rs/stm32f4xx-hal #61: rcc freeze() panics with PR #59

arkap, 2019-01-16:
I'm using this crate through the [STM32f407g-discovery board-support crate](https://github.com/therealprof/stm32f407g-disc). I noticed that ever since #59 got merged, `rcc.freeze()` panics because [this assertion](https://github.com/stm32-rs/stm32f4xx-hal/blob/master/src/rcc.rs#L232) fails. In my code `pclk2_max == 90_000_000` and `pclk2 == 168_000_000` (which is the system clock).

therealprof, 2019-01-16:
Sounds like multiple bugs to me: PCLK2 on the STM32F407 is not allowed to by higher than 84MHz (not 90MHz).

@aurelj Can you have a look?

aurelj, 2019-01-17:
@therealprof, you're right, we need to constrain PCLK2 to 84 MHz instead of 90 MHz on the STM32f407.
I propose a fix in  #63.

@arkap if you set a value for pclk2 in your code, it must respect the constraint specified in the Reference Manual of your STM32F407, so in practice, you can't set your pclk2 above 84 MHz. You can try for example:
```
rcc.cfgr.sysclk(168.mhz()).pclk2(84.mhz()).freeze();
```