stm32-rs/stm32f4xx-hal #62: How to access the status register (sr) when using the Serial hal implementation?

timbod7, 2019-01-16:
I'm (finally) trying to make use of this feature:

https://github.com/stm32-rs/stm32f4xx-hal/pull/42

In order to determine what caused the usart interrupt, I need to access the usart status register (sr) from the interrupt handler. But with code like this to initialize the usart:

```
        let config = serial::config::Config::default()
            .baudrate(100_000.bps())
            .parity_even()
            .wordlength_9()
            .stopbits(serial::config::StopBits::STOP2);
        let serial = Serial::usart2(p.USART2, (txpin, rxpin), config, clocks).unwrap();
        let (mut tx, mut rx) = serial.split();
```

the move semantics for the first parameters of `Serial::usart2()` means that I cannot subsequently access `p.USART2.sr`.

Can you suggest a way to access `sr` after calling `serial.split()` ?

therealprof, 2019-01-16:
Do you need to use `split()`?

timbod7, 2019-01-16:
I thought that `split()` was the means by which I get access to the `hal::serial::Read` and `hal::serial::Write` abstractions.

In any case, even without `split()` the move semantics of `Serial::usart2()` means that I still couldn't access  `p.USART2.sr`.

timbod7, 2019-01-16:
Whilst my use case for  idle events is perhaps unusual, I think I would have the same issue if I had enabled both `Txe` and `Rxne` events and needed to distinguish between them in the interrupt handler.

psycowithespn, 2019-01-16:
A trick I used for the rxne and txe interrupts is calling the read and
write methods. If it returns a value, you have it or it gets sent.
Otherwise, it returns WouldBlock, indicating it wasn't that interrupt. This
won't help with idle, though.

timbod7, 2019-01-16:
@psycowithespn: yes - though relying on non blocking reads to detect flag state does feels like a "trick" rather than a clean solution.

therealprof, 2019-01-16:
@timbod7 

> I thought that split() was the means by which I get access to the hal::serial::Read and hal::serial::Write abstractions.

Nah, we're dialling that back in bit in various places. I see the split function mostly useful if you want to do Rx/Tx in completely different places (e.g. share one USART for 2 unidirectional communication channels). If you want to do general bidirectional communication I'd just use the Serial directly (NB: I haven't verified that Read/Write are actually implemented on Serial but I thought they were; if not we should change that).

I'd also say that we should add a function to `Serial` and `Tx`/`Rx` to retrieve the current status. We could even do it as a free function if you only need RO access to it. For clearing you'd need to use `Serial` though.

timbod7, 2019-01-16:
> I haven't verified that Read/Write are actually implemented on Serial but I thought they were; if not we should change that

It doesn't. I can put up a PR for this?

therealprof, 2019-01-16:
Yes please.

timbod7, 2019-01-17:
I was incorrect on this. `Serial` already implements the `Read` and `Write` traits. But I will still put together a PR for an accessor for the status register.

timbod7, 2019-01-20:
https://github.com/stm32-rs/stm32f4xx-hal/pull/65

aurelj, 2019-02-09:
#65 was merged and I think it fixes this issue so closing now.

closed 2019-02-09 by aurelj