stm32-rs/stm32f4xx-hal #67: Add PWM support

nordmoen, 2019-01-28:
It would be nice to support PWM ala: https://github.com/japaric/stm32f103xx-hal/blob/master/src/pwm.rs .

A first approach might be to simply allow `PXx` to go into [`Alternate<PushPull>` mode](https://github.com/japaric/stm32f103xx-hal/blob/master/src/gpio.rs#L186) and let users set up timing (which is possible today).

therealprof, 2019-01-28:
@nordmoen I also started work on a PWM implementation for STM32F0xx here: https://github.com/stm32-rs/stm32f0xx-hal/blob/features/pwm/src/pwm.rs

I works in principle but is still awaiting an API overhaul and generalisation. I haven't checked for PWM but in other instances F4 peripherals operation are closer to F0 than F1...

nordmoen, 2019-01-28:
Thanks @therealprof I'll check it out once my new development board arrive.

danclive, 2020-01-15:
How is it going？I met the same need.

closed 2020-04-29 by therealprof