stm32-rs/stm32f4xx-hal #75: Update to stm32f4-0.6.0

tdaede, 2019-02-14:
Unfortunately having any mismatch here causes two versions of stm32f4 to be built, and a linker error:

  = note: rust-lld: error: duplicate symbol: DEVICE_PERIPHERALS
          >>> defined at mod.rs:1240 (/home/thomas/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32f4-0.4.0/src/stm32f401/mod.rs:1240)
          >>>            stm32f4-8d4128126bbacf9e.stm32f4.ahbyhsiv-cgu.8.rcgu.o:(DEVICE_PERIPHERALS) in archive /home/thomas/sandbox/td-crt/td-deflect/td-deflect-firmware/target/thumbv7em-none-eabihf/debug/deps/libstm32f4-8d4128126bbacf9e.rlib
          >>> defined at mod.rs:1212 (/home/thomas/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32f4-0.6.0/src/stm32f401/mod.rs:1212)
          >>>            stm32f4-102ad208fe39464e.stm32f4.sgay4gpn-cgu.9.rcgu.o:(.bss.DEVICE_PERIPHERALS+0x0) in archive /home/thomas/sandbox/td-crt/td-deflect/td-deflect-firmware/target/thumbv7em-none-eabihf/debug/deps/libstm32f4-102ad208fe39464e.rlib

therealprof, 2019-02-14:
Not sure I follow. Why do you link against two different versions?

tdaede, 2019-02-14:
I want to use newer bits from the stm32f4 crate but stm32f4xx-hal also depends on stm32f4, so cargo installs 0.4.0 for it.

(unless there is something I'm missing here?)

therealprof, 2019-02-14:
So in a nutshell you'd like use to do a release?

tdaede, 2019-02-14:
Aha, swapped to git = "" and it works.

Yes :)

commit 9cbdfa72 (2019-02-14): Merge pull request #76 from stm32-rs/release-0.3.0

Bumped version to 0.3.0, Closes #75
closed 2019-02-14 by therealprof