stm32-rs/stm32f4xx-hal #77: External interrupt source not correct

kitzin, 2019-02-27:
Relevant code and relevant datasheet section from `STM32F411RE` MCU.
https://github.com/stm32-rs/stm32f4xx-hal/blob/216f7f62e2b7d035b10a9e060a10de44faff6ce5/src/gpio.rs#L190-L217
https://github.com/stm32-rs/stm32f4xx-hal/blob/216f7f62e2b7d035b10a9e060a10de44faff6ce5/src/gpio.rs#L89-L92
https://github.com/stm32-rs/stm32f4xx-hal/blob/216f7f62e2b7d035b10a9e060a10de44faff6ce5/src/gpio.rs#L713-L730
[image]

So for example if I would to be pin `PB2` the interrupt line it should select is `EXTI2`.

```
$extigpionr = 1
$i = 1
```
which would imply
```
offset = 4
self.i = 1
```
in `make_interrupt_source`

This writes `0bXXXX_XXXX_0001_XXXX` to the `SYSCFG_EXTICR1` register. This enables `EXTI1` which is not the correct one for `PB2` pin.

Not completely sure if I missed something, or the implementation is not correct?

Thanks!

aurelj, 2019-02-27:
Indeed, the implementation looks wrong.
offset should be:
`let offset = 4 * (self.i % 4);`

A fix is on its way.

aurelj, 2019-02-27:
Should be fixed now.
Oh, and thanks for the detailed bug report !

PR #78 (2019-02-27): gpio: fix exticr offset to match the pin num rather than the port num
fix #77
commit 106b4743 (2019-02-27): gpio: fix exticr offset to match the pin num rather than the port num

fix #77
closed 2019-02-27 by aurelj