stm32-rs/stm32f4xx-hal #79: Pll config invalid for 8mHz HSE

cs2dsb, 2019-03-02:
Hey,

In rcc pll config there is the following line:
```
// Input divisor from PLL source clock, must result to frequency in the range from 1 to 2 MHz
let pllm = 16;
```
If your HSE is 8mhz this results in a pll source clock of 500_000 which isn't valid, as the comment says.

Pull request #57 appears to fix this among other things.

It's a shame the writeConstraints in the SVD don't propagate through and enforce valid values for the fields - the ranges are all set correctly by stm32-rs but svd2rust just marks the fields as unsafe if the range doesn't cover the full width of the fields. As a fallback position, would it be a good idea to put some asserts or debug assertions in this crate to enforce valid values in the fields?

Cheers

cs2dsb, 2019-03-02:
After a bit more testing, #57 doesn't fix it entirely. It works with the 8mhz crystal if  you want a target clock of 50mhz but 100mhz results in a plln of 800 which is invalid.

Also, I noticed the following in the reference manual:
```
PLLM:
It is recommended to select a frequency of 2 MHz to limit PLL jitter.
```

Changing #57's pll setup code set pllm to 2mhz also allows it to correctly set the parameters to achieve a 100mhz sysclk with an 8mhz hse.

aurelj, 2019-03-03:
Thanks for the report.
PR #80 should fix this issue.

PR #80 (2019-03-03): rcc: support PLL configuration for HSE < 16 MHz
fix #79
commit a9b4037a (2019-03-03): rcc: support PLL configuration for HSE < 16 MHz

fix #79
closed 2019-03-03 by aurelj