stm32-rs/stm32f4xx-hal #85: Cannot use USART 1 of STMF407VG (discovery)

paulvt, 2019-03-13:
I have tried to setup and use the USART 1 for serial communication. If the baud rate is 1200/2400, it works; around 4800 communication becomes corrupted and 19k2 or above just doesn't seem to work anymore.  I have adapted the "serial_echo" example from the stm32f407g-disc create to work on this HAL crate as follows:

```rust
#![no_main]
#![no_std]

extern crate cortex_m_rt;
extern crate panic_semihosting;

use cortex_m_rt::entry;
use hal::block;
use hal::prelude::*;
use hal::serial::{config::Config, Serial};

#[entry]
fn main() -> ! 
    if let Some(p) = hal::stm32::Peripherals::take() {
        let gpioa = p.GPIOA.split();
        let rcc = p.RCC.constrain();
        let clocks = rcc.cfgr.sysclk(48.mhz()).freeze();

        // USART 1 at PA9 (TX) and PA10(RX)
        let tx = gpioa.pa2.into_alternate_af7();
        let rx = gpioa.pa3.into_alternate_af7();

        // Set up USART 1 configured pins and a baudrate of 115200 baud
        let serial = Serial::usart1(
            p.USART1,
            (tx, rx),
            Config::default().baudrate(115_200.bps()),
            clocks,
        )
        .unwrap();

        // Separate out the sender and receiver of the serial port
        let (mut tx, mut rx) = serial.split();

        loop {
            // Read character and echo it back
            let received = block!(rx.read()).unwrap();
            block!(tx.write(received)).ok();
        }
    }

    loop {
        continue;
    }
}
```

But this results in a panic, even though we are only configuring the sysclk:
```
panicked at 'assertion failed: pclk1 <= pclk1_max', /home/paul/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32f4xx-hal-0.3.0/src/rcc.rs:277:9
```

When leaving out the `sysclk(48.mhz())` call, it compiles, but communication seems inpossible.
When setting to 9600 baud and sending 0xAA (10101010), I get back 0xFA or 0xEA.

See also the attached output below of a logic analysis where you can see that the TX part has malformed with respect to what was sent on RX.

[image]

When changing the code to use USART 2 (pins PA1/PA2) instead, everything works just fine.

paulvt, 2019-03-13:
Note that I am really new (as in, just two weeks in) to embedded development :smile:.

aurelj, 2019-03-14:
Please try the fix in PR #86.
This should allow you to set sysclk to 48 MHz.

Without this fix, you can also try to set your sysclk to 42 MHz or 84 MHz instead.

paulvt, 2019-03-15:
I can confirm that I can set the sysclk to 48MHz with this fix, but bytes still are not echoed correctly unfortunately.

aurelj, 2019-03-15:
Ok. You may also try higher clock speed (STM32F407 support 168 MHz max).

But the issue may also be in the serial baudrate / divider... Or totally unrelated.

paulvt, 2019-03-15:
That also doesn't help unfortunately. So it doesn't seem to be sysclk-related then?

aurelj, 2019-03-16:
Probably not.
The issue in your logic analyzer output TX is the inconsistency in the duration of bits. Look at the length of the start bit (0) and the last data bit just before stop (also 0). They are exactly the same duration in the RX trace but they have a quite different duration in the TX trace.
I'm not sure what could explain this inconsistency. Maybe something in the uart configuration (in the serial module) but I don't wee what. Maybe a hardware issue ? High capacitance on your line ? An oscilloscope would help to check the rise and fall time on your lines.

paulvt, 2019-03-16:
Ah yes, a very similarly thing was mentioned to me. I will put it under a scope and report back here.

therealprof, 2019-03-16:
@paulvt So I was just trying your example from above... It does not even compile but I noticed some discrepancies: Are you trying to use PA9 and PA10 without hardware modifications? That's likely not going to work because they are used for USB stuff.

therealprof, 2019-03-16:
So, just to rule out a hardware conflict in your constellation and a bug in the use of the clocks/baudrate calculation I tried USART1 with PB6/PB7 and TX works just fine at 115kbaud with both 42Mhz and 48Mhz SysClk.

paulvt, 2019-03-16:
I have fixed the example above; for some reason it still used PA2/PA3 (in contrary to the comments) which was in the example from the board crate that used USART 2. That explains why it didn't compile for you.

So yes, I am using pins PA9/PA10 and without hardware modifications. I do not recall which of the dozens of documents out there pointed me to those pins and hadn't even realized that PB6/PB7 could be used as well (and without the USB conflict)?!  So maybe the only bug here is the one where 48MHz could not be configured, fixed by PR #86?

therealprof, 2019-03-16:
Everything seems to work fine for me with the latest master. I'd highly recommend you try STM32CubeMX to support your discoveries (yeah, sorry for the pun 😅).

therealprof, 2019-03-17:
@paulvt FYI I just updated the `stm32f407g-disc` crate and moved it to the stm32-rs repository.

Please let us know whether this is finally resolved and can be closed.

paulvt, 2019-03-20:
I just confirmed that with git master of `stm32f4xx-hal` USART 1 works fine provided PB6 and PB7 are used.

PR #86 (2019-03-15): rcc: fix rounding of hclk and pclk divisor
fix #85
closed 2019-03-20 by paulvt