stm32-rs/stm32f4xx-hal #99: The I2c driver does not check/set open_drain

korken89, 2019-05-21:
Hi,

I was having issues with the I2c and it turned out to be that I had not done `set_open_drain()` on the pins.

There should be a check, or that the I2c should set this - as missing this does not produce a compile error and it can in worst case burn a slave when the master holds the bus high and the slave wants to ACK the address.
As it is now should be considered a design error as it can cause irreparable damage to a system without (compile-time/run-time) warning.

BR Emil

therealprof, 2019-05-21:
There're many ways to destroy your hardware by improper use, even by use of software and there's nothing one can do about it. In this particular case we could probably address it and I wouldn't object to having a compile time error if someone doesn't put the pin in the right state for I2C operation.

Any takers to create a PR? 😉

VersBinarii, 2019-08-27:
I would love to help with this but i would be grateful for some guidance as i'm not exactly sure how to implement this as a compile-time check.

therealprof, 2019-08-28:
@VersBinarii There're a few different ways to address this. I haven't looked into the details so I'm just citing from the top of my head.

- First idea: We're already using type states to make sure that when we hand a GPIO pin over to the I2C initialisation it is in the correct mode. One of the possible ways to address this would be to extend the type states to not only include the alternate function assignment but also the pin drive mode.
- Second idea: We could simply initialise the pin correctly in the I2C initialisation, since we have full control over the pin at that point and other drive modes do make make too much sense for I2C operation.

So the first idea would be the option where the compiler checks the state for us, disadvantage: More fiddly code to handle for the maintainers of the crate. The second idea is where we simply do the right thing(tm) behind the users back, disadvantage: Might require more input parameters to implement safely.

VersBinarii, 2019-08-29:
@therealprof Yes, i though about the second approach as well because it seemed to me that the first approach would become a bit messy. I also didn't see this being done in any other HALs i saw.

I just thought that the compile time check would be more Rusty and therefore preferable.

bfrog, 2020-02-06:
It seems like if your are setting the pin for an alternative function (i2c) the crate should appropriately setup the pin drive mode for you for that alternative function with a sensible default. I ran into this myself, and its less than obvious from the API that it's up to me to set that up appropriately.

Saying that I think the second option would be my preferred take on this.

therealprof, 2020-02-06:
@bfrog Mind trying a PR to implement this?

bfrog, 2020-02-07:
@therealprof I'll try to find some time tomorrow to do this

therealprof, 2020-02-07:
Great.

bfrog, 2020-02-08:
@therealprof I promised to look and I did, the types PinScl and PinSda would I think need to expose some stuff for each impl which is quite a lot of effort and code for what amounts to what I was hoping was just a function call away, macros or more traits could solve it but I ran out of time

therealprof, 2020-02-08:
Yeah, type states can be very annoying. Also the relevant functions consume the pin so I guess we'll need to check for another approach.

bfrog, 2020-02-08:
To make it easier each pin type could implement a common trait that exposes set_open_drain(). This could let PinScl PinSda, and the Pins traits in i2c have a default trait func that configures them appropriately. That seems like an ok way of doing it to me, but seemed to go against the macro flow going on.

therealprof, 2020-02-08:
Macro flow?

I was thinking of introducing a new I2C mode which automatically does the right thing and deprecate the old AF4/AF9 modes.

PR #122 (2020-02-09): Added open drain Alternate modes and ways to construct them
This allows us to differentiate between a non open-drain and open-drain
alternate function configuration and require the open-drain variant for
I2C use to prevent hot smoke resulting from a misconfiguration.

Fixes #99
commit b6f01053 (2020-02-09): Added open drain Alternate modes and ways to construct them

This allows us to differentiate between a non open-drain and open-drain
alternate function configuration and require the open-drain variant for
I2C use to prevent hot smoke resulting from a misconfiguration.

Fixes #99
closed 2020-02-09 by therealprof