stm32-rs/stm32f4xx-hal #101: Slightly incorrect PLL divisions when HSE is odd

MabezDev, 2019-07-17:
When using the 16mhz HSI, the pll divisions are perfect. When using the HSE with a 25mhz crystal the generated clocks are close, but not close enough.

The cfgr configuration:

```
let clocks = rcc.cfgr
                            .use_hse(25.mhz())
                            .sysclk(120.mhz())
                            .pclk1(30.mhz())
                            .pclk2(60.mhz())
                            .use_pll48mhz()
                            .freeze();
```

HSI(16mhz):

```
INFO  src/main.rs:99 - Clocks {
    hclk: Hertz(
        120000000,
    ),
    pclk1: Hertz(
        30000000,
    ),
    pclk2: Hertz(
        60000000,
    ),
    ppre1: 4,
    ppre2: 2,
    sysclk: Hertz(
        120000000,
    ),
}
```

HSE (25mhz):

```
INFO  src/main.rs:99 - Clocks {
    hclk: Hertz(
        119230712,
    ),
    pclk1: Hertz(
        29807678,
    ),
    pclk2: Hertz(
        59615356,
    ),
    ppre1: 4,
    ppre2: 2,
    sysclk: Hertz(
        119230712,
    ),
}
```

A valid clock configuration can be calculated in CubeMX (this is an attempt to port an existing project to Rust), just wondering if its possible to adjust the calculations such that the HAL can find them too?

therealprof, 2019-07-17:
Sure, if someone provides a PR with the correct calculations we can apply it.

aurelj, 2019-07-17:
Please try patch from PR #102, this is a quick try at fixing this issue.

PR #102 (2019-07-18): rcc: find best pllm value to get sysclk as close as possible to requested value
Fixes #101
commit 44ec5ab4 (2019-07-18): rcc: find best pllm value to get sysclk as close as possible to requested value

Fixes #101
closed 2019-07-18 by therealprof