stm32-rs/stm32f4xx-hal #106: Add Timer::stop

PvdBerg1998, 2019-07-21:
Please expose a way to stop a timer. Even better would be a way to init a timer without starting it.
The pause line, located at https://docs.rs/stm32f4xx-hal/0.5.0/src/stm32f4xx_hal/timer.rs.html#210-231 could be extracted if I'm correct. The stm32f1xx hal (originally by japaric) already does this: https://docs.rs/stm32f1xx-hal/0.3.0/src/stm32f1xx_hal/timer.rs.html#149.

MarcoIeni, 2020-03-25:
I noticed that the `Cancel` trait exists in the [timer](https://docs.rs/crate/embedded-hal/0.2.3/source/src/timer.rs) API. Maybe for this function we should implement it?

therealprof, 2020-03-26:
@MarcoIeni Sounds like a good idea. Want to give it a shot?

MarcoIeni, 2020-03-26:
Yes, I will try to work on this :)

MarcoIeni, 2020-05-01:
A way to stop a timer has been provided by #137 and #148.

> Even better would be a way to init a timer without starting it.

Should we do this? Otherwise we can close this issue.

closed 2024-06-01 by burrbull