stm32-rs/stm32f4xx-hal #107: linking with `rust-lld` failed when stm32f4 crate is 0.8.0

btrepp, 2019-08-26:
So I was playing with this crate, and it seems only compatible with stm32f4 0.7.0,
if you have 0.8.0 in the Cargo.toml file, you get the below error

> error: linking with `rust-lld` failed: exit code: 1                                                                                                                                                                                                                                                                                                                                                                                                                                             |                                                                                                                                                                                                                                                                                                                                                                                                                                                                                             = note: "rust-lld" "-flavor" "gnu" "-L" "C:\\Users\\Beau\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rus                                                                                                                                                                                                                                                                                                                                                                      tlib\\thumbv7em-none-eabihf\\lib" "C:\\Users\\Beau\\Code\\rust\\nucleo\\target\\thumbv7em-none-eabihf\\debug\\deps\\nucl                                                                                                                                                                                                                                                                                                                                                                      eo-eaaf61e3bc003baa.271x5gr5qamfyx15.rcgu.o" "C:\\Users\\Beau\\Code\\rust\\nucleo\\target\\thumbv7em-none-eabihf\\debug\                                                                                                                                                                                                                                                                                                                                                                      \deps\\nucleo-eaaf61e3bc003baa.2rkr7foartczqb4x.rcgu.o" "C:\\Users\\Beau\\Code\\rust\\nucleo\\target\\thumbv7em-none-eab                                                                                                                                                                                                                                                                                                                                                                      ihf\\debug\\deps\\nucleo-eaaf61e3bc003baa.2ual8ehbeivzvzip.rcgu.o" "C:\\Users\\Beau\\Code\\rust\\nucleo\\target\\thumbv7                                                                                                                                                                                                                                                                                                                                                                      em-none-eabihf\\debug\\deps\\nucleo-eaaf61e3bc003baa.35u4h8fbwv4k3f1.rcgu.o" "C:\\Users\\Beau\\Code\\rust\\nucleo\\targe                                                                                                                                                                                                                                                                                                                                                                      t\\thumbv7em-none-eabihf\\debug\\deps\\nucleo-eaaf61e3bc003baa.3giu8u2ftlj6pjgq.rcgu.o" "C:\\Users\\Beau\\Code\\rust\\nu                                                                                                                                                                                                                                                                                                                                                                      cleo\\target\\thumbv7em-none-eabihf\\debug\\deps\\nucleo-eaaf61e3bc003baa.3kxoq02mgsdnfnvw.rcgu.o" "C:\\Users\\Beau\\Cod                                                                                                                                                                                                                                                                                                                                                                      e\\rust\\nucleo\\target\\thumbv7em-none-eabihf\\debug\\deps\\nucleo-eaaf61e3bc003baa.46wer6d7gc7foq85.rcgu.o" "C:\\Users                                                                                                                                                                                                                                                                                                                                                                      \\Beau\\Code\\rust\\nucleo\\target\\thumbv7em-none-eabihf\\debug\\deps\\nucleo-eaaf61e3bc003baa.4pd5vo7t97ol8azi.rcgu.o"                                                                                                                                                                                                                                                                                                                                                                       "C:\\Users\\Beau\\Code\\rust\\nucleo\\target\\thumbv7em-none-eabihf\\debug\\deps\\nucleo-eaaf61e3bc003baa.4vxh1s0pjgr87                                                                                                                                                                                                                                                                                                                                                                      h8.rcgu.o" "C:\\Users\\Beau\\Code\\rust\\nucleo\\target\\thumbv7em-none-eabihf\\debug\\deps\\nucleo-eaaf61e3bc003baa.4yx                                                                                                                                                                                                                                                                                                                                                                      wb4agdbctcsoq.rcgu.o" "C:\\Users\\Beau\\Code\\rust\\nucleo\\target\\thumbv7em-none-eabihf\\debug\\deps\\nucleo-eaaf61e3b                                                                                                                                                                                                                                                                                                                                                                      c003baa.50bp00c2ip9fgqaq.rcgu.o" "C:\\Users\\Beau\\Code\\rust\\nucleo\\target\\thumbv7em-none-eabihf\\debug\\deps\\nucle                                                                                                                                                                                                                                                                                                                                                                      o-eaaf61e3bc003baa.5361pmkaifxg2t6q.rcgu.o" "C:\\Users\\Beau\\Code\\rust\\nucleo\\target\\thumbv7em-none-eabihf\\debug\\                                                                                                                                                                                                                                                                                                                                                                      deps\\nucleo-eaaf61e3bc003baa.5cc3h2ivztv8wejj.rcgu.o" "C:\\Users\\Beau\\Code\\rust\\nucleo\\target\\thumbv7em-none-eabi                                                                                                       [... cut ...]                                                                                                                                                                                                                                                                                                                                                                                                                                                                 To learn more, run the command again with --verbose.

btrepp, 2019-08-26:
I seem to be able to get it to build if I modify it slightly. I will raise a PR

aurelj, 2019-10-03:
Fixed by the merged PR #108

closed 2019-10-03 by aurelj