stm32-rs/stm32f4xx-hal #109: Wrong max clock on F407?

wjakobczyk, 2019-10-03:
Hi,

I have observed slower performance of my C code when executed from Rust, compared to when executing from C app. I noticed different clock register values in both configurations.
In Rust I setup my clock as either:
`rcc.cfgr.sysclk(stm32f4xx_hal::time::MegaHertz(168)).freeze();`
or:
```
rcc.cfgr
                            .use_hse(stm32f4xx_hal::time::MegaHertz(8))
                            .sysclk(stm32f4xx_hal::time::MegaHertz(168))
                            .pclk1(stm32f4xx_hal::time::MegaHertz(42))
                            .pclk2(stm32f4xx_hal::time::MegaHertz(84))
                            .freeze();
```
(no difference in observed performance).
In C config, system clock is setup by ST-provided code:
  ```
*=============================================================================
  *        Supported STM32F4xx device revision    | Rev A
  *-----------------------------------------------------------------------------
  *        System Clock source                    | PLL (HSE)
  *-----------------------------------------------------------------------------
  *        SYSCLK(Hz)                             | 168000000
  *-----------------------------------------------------------------------------
  *        HCLK(Hz)                               | 168000000
  *-----------------------------------------------------------------------------
  *        AHB Prescaler                          | 1
  *-----------------------------------------------------------------------------
  *        APB1 Prescaler                         | 4
  *-----------------------------------------------------------------------------
  *        APB2 Prescaler                         | 2
  *-----------------------------------------------------------------------------
  *        HSE Frequency(Hz)                      | 8000000
  *-----------------------------------------------------------------------------
  *        PLL_M                                  | 8
  *-----------------------------------------------------------------------------
  *        PLL_N                                  | 336
  *-----------------------------------------------------------------------------
  *        PLL_P                                  | 2
  *-----------------------------------------------------------------------------
  *        PLL_Q                                  | 7
  *-----------------------------------------------------------------------------

```
When I dump RCC register they do differ.
In Rust:
(gdb) x/16x 0x40023800
```
0x40023800:	0x03038583	0x24402a04	0x0000940a	0x00000000
0x40023810:	0x00000000	0x00000000	0x00000000	0x00000000
0x40023820:	0x00000000	0x00000000	0x00000000	0x00000000
0x40023830:	0x00100008	0x00000000	0x00000000	0x00000000

```
In C:
(gdb) x/16xw 0x40023800
0x40023800:	0x03038583	0x07405408	0x0000940a	0x00000000
0x40023810:	0x00000000	0x00000000	0x00000000	0x00000000
0x40023820:	0x00000000	0x00000000	0x00000000	0x00000000
0x40023830:	0x00100008	0x00000000	0x00000000	0x00000000

How can I achieve the same configuration, and performance, in Rust?

aurelj, 2019-10-03:
According to your RCC register dump, here are the PLL values set by the Rust version:
```
PLLM = 4
PLLN = 168
PLLP = 2
PLLQ = 4
```
Those different parameters should generate the same 168 MHz sysclk as the C version, but with a different VCO input (2 MHz for the Rust version as recommended in Reference Manual versus 1 MHz for the C version).
You can also check the content of the Clocks struct returned by .freeze() to verify if the values are OK.

The Rust version should run at the same frequencies as the C version, so if you have a performance difference it might come from something else.

wjakobczyk, 2019-10-04:
There is also a difference in PLLQ (value 7 in the C version), but it should not have impact on CPU.
And, there are differences in the reserved bits, we should be left intact according to the reference manual.
To my Rust version I have added a call to a C function that sets up the clocks in the C configuration and then performance is ok. So there must be something wrong with the clocks setup by Rust code. I'll keep digging and let you know.

therealprof, 2019-10-04:
> And, there are differences in the reserved bits, which should be left intact according to the reference manual.

Reserved bits will often return random data upon read.

Are you sure it's not something else like the flash access wait states? I confirmed a while ago using MCO and an oscilloscope that the clocks are set up okay.

wjakobczyk, 2019-10-04:
Actually, the C setup function does configure flash:

```
    /* Configure Flash prefetch, Instruction cache, Data cache and wait state */
    FLASH->ACR = FLASH_ACR_ICEN |FLASH_ACR_DCEN |FLASH_ACR_LATENCY_5WS;

```
I will check it  later today.

wjakobczyk, 2019-10-04:
Confirmed, it's the flash settings that make the difference.
Specifically, enabling of data cache and instruction cache in the FLASH_ACR register.

aurelj, 2019-10-05:
@wjakobczyk, could you try the patch in PR #110 to confirm that it fixes the speed difference for you ?

closed 2019-10-06 by therealprof

wjakobczyk, 2019-10-06:
Yes, the merged patch solves the problem, thanks!

PR #110 (2019-10-06): enable instruction and data cache and instruction prefetch
this should fix #109
commit 344d0a37 (2019-10-06): enable instruction and data cache and instruction prefetch

this should fix #109