stm32-rs/stm32f4xx-hal #115: example stopwatch-with-ssd1306-and-interrupts fails to build

pdgilbert, 2019-11-06:
$ cargo build  --target thumbv7em-none-eabihf  --features stm32f411,rt --example stopwatch-with-ssd1306-and-interrupts
   Compiling stm32f4xx-hal v0.6.0 (/home/paul/rust/egAttempts/stm32f4xx-hal)
error[E0061]: this function takes 0 parameters but 1 parameter was supplied
   --> examples/stopwatch-with-ssd1306-and-interrupts.rs:149:17
    |
149 |             btn.clear_interrupt_pending_bit(exti);
    |                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^ expected 0 parameters

error: aborting due to previous error

For more information about this error, try `rustc --explain E0061`.
error: Could not compile `stm32f4xx-hal`.

therealprof, 2019-11-06:
I guess we deserve that for not running our (pretty new) examples through the CI... 😅

therealprof, 2019-11-06:
Fixed by: 231941f53a45176eea472663a1ec586b5d1119f4

closed 2019-11-06 by therealprof