stm32-rs/stm32f4xx-hal #117: memory.x wrong for stopwatch-with-ssd1306-and-interrupts

pdgilbert, 2019-11-06:
(somewhat related to #115 ) Not sure if you deserve this,  but maybe you are not running  CI with the git memory.x ?
With memory.x from the git repository, which has
  FLASH : ORIGIN = 0x08000000, LENGTH = 64K 
  RAM : ORIGIN = 0x20000000, LENGTH = 32K

For 'cargo build  --target thumbv7em-none-eabihf  --features stm32f411,rt --example  stopwatch-with-ssd1306-and-interrupts'  I get  
= note: rust-lld: error: section '.rodata' will not fit in region 'FLASH': overflowed by 12 bytes
          rust-lld: error: section '.rodata' will not fit in region 'FLASH': overflowed by 49 bytes
...
error: Could not compile `stm32f4xx-hal`.

There is no problem with the memory.x for my actual device, which has 
  /* Define memory regions.  STM32F411RE */
  FLASH (rx) : ORIGIN = 0x08000000, LENGTH = 512K
  RAM (rwx)  : ORIGIN = 0x20000000, LENGTH = 128K

therealprof, 2019-11-06:
Yeah, the settings are very conservative so they work with the smallest MCUs from the F4 family. If you compile with `--release` it's not an issue.

therealprof, 2019-11-06:
The conservative settings are on purpose so I'm closing this. I'd recommend creating your own workspace and just depend on this crate, then you can have your own meaningful configuration for the actual MCU you have.

closed 2019-11-06 by therealprof