stm32-rs/stm32f4xx-hal #119: RCC constrain eats the RCC

diondokter, 2019-11-20:
When calling the constrain function on the RCC, it eats the register. This is highly inconvenient as I want to later enable and disable different peripheral clocks and can't access the register. Now I need to resort to getting the register unsafely which I want to minimize.

Please consider to either give it back when freeze is called or implement the turning on and off of peripheral clocks in the Rcc type.

therealprof, 2019-11-20:
> When calling the constrain function on the RCC, it eats the register. 

Yes, the register is moved into the the `Rcc` type. And yes, that's on purpose for safety reasons. If you want to enable peripheral clocks, we could add methods on the `Rcc` to allow doing that.

maxekman, 2020-04-27:
See this blog post by @japaric for more info about how and why: https://blog.japaric.io/brave-new-io/#freezing-the-clock-configuration

diondokter, 2020-04-27:
> 
> 
> See this blog post by @japaric for more info about how and why: https://blog.japaric.io/brave-new-io/#freezing-the-clock-configuration

I agree with the principle, but not all the peripherals are there. When I made this issue, I had a need of turning the CAN clock on. When all the options are accounted for, then it's a great idea!

Also, something else to consider:
When designing a low power battery device, you may want to turn off some clocks to save power. That also isn't possible right now without major work.

maxekman, 2020-04-27:
I share the same situation as you; in my case I’m working on enabling the I2S clocks for a HAL proposal. My current experimentation with this crate is similar to a combination of how the peripheral clocks and 48MHz (USB I think) clock is configured. I have added a function called `plli2sclk(freq/audio sampling rate)`. I think that is a pretty decent way to go for now, with a balance of safety and configurability.

muttering-oldman, 2020-08-11:
I have the same difficulties as the **diondokter**. I need to enable RTC, but "RTC.constrains" eats the register.
I'm new to rust embedded and I do a lot of things by examples. Unfortunately, I have not found the RTC example.
I tried to do it by datasheet but came across that the register doesn't live up to the moment I need it.
I have to get to the bdcr register to enable lse.
It would be great to add work with RTC like in stm32f1xx-hal.

birkenfeld, 2020-08-11:
@muttering-oldman note that you can always bypass the HAL by using e.g. `unsafe { *RTC::ptr() }`.  This can be necessary in cases where the HAL doesn't expose the functionality you need.

muttering-oldman, 2020-08-11:
Thank you @birkenfeld.
I will try to bypass HAL :)
But still it would be nice to have this feature out of the box.

ryan-summers, 2020-08-15:
In the STM32H7xx-hal, we get around this by having the RCC constrain function return a PREC structure, which essentially provides a proxy into the RCC to enable and reset various peripherals.

Reference code: https://github.com/stm32-rs/stm32h7xx-hal/blob/master/src/rcc/rec.rs

therealprof, 2020-08-15:
I agree that having some proxy is probably the way to go.