stm32-rs/stm32f4xx-hal #128: Looking for a working f4 hal i2c read example

tstellanova, 2020-02-25:
I've tried a couple different embedded rust projects that use embedded hal i2c Read (or WriteRead) to read data from a sensor, with no success.  Is there an i2c Read example that's known to work with a common F4 dev board?  

---
Details:
I've tried using the  STM32F4Discovery board and cheap stm32f401CxUx board.  I've tried both a sensor driver that uses WriteRead and one that uses Read.  I've tried two distinctly different sensors from two different manufacturers. 

WriteRead appears to work once or twice but then subsequently hangs in eg `i2c::write`:

```
        // Wait until START condition was generated
        while self.i2c.sr1.read().sb().bit_is_clear() {}
```

therealprof, 2020-02-25:
Hm, weird. No pull-ups? I haven't tried reading but writing works quite okay, as exhibited by the included examples.

tstellanova, 2020-02-26:
Agreed, writing seems to work fine, but Read and WriteRead seem to have problems. 
Looking at stm32f3xx-hal and stm32h7xx-hal, they seem to use different code for Read. I’ll try to setup a test with h7 or f3 and see if I have any better luck.

tstellanova, 2020-02-27:
Starting to narrow down the issue.  It appears the problem is with multi-byte reads.  The usage pattern is:
- Write one byte (for some drivers this selects a register or subaddress)
- Read multiple bytes

I verified that single-byte WriteReads (write one byte, read one byte) work fine for thousands of cycles with stm32f4xx-hal on stm32f401CxUx.  As soon as I introduce a multi-byte WriteRead (write one, read 18), the program hangs waiting to obtain master status. 

I verified that these same multi-byte WriteReads work fine with stm32f334disco and stm32f3xx.  That hal uses a completely different WriteRead implementation. 

FWIW here's the [bare metal app](https://github.com/tstellanova/test_em7180) I'm using to test this. It requires an EM7180 sensor / [USFS AHRS](https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution-lsm6dsm-lis2md/)   I'm trying to locate a simpler embedded hal driver that carries out multi byte i2c reads.

tstellanova, 2020-02-27:
Looks like the bmp280 driver does [multi-byte WriteRead](https://github.com/copterust/bmp280/blob/master/src/lib.rs#L71). 
 I'll give this one a try as well.

tstellanova, 2020-02-27:
I've boiled down a reproducible test case with a bare metal app that talks to the widely available bmp280 barometer [here](https://github.com/tstellanova/test_bmp280).  This sensor driver uses multi-byte WriteRead with i2c. 

This example works fine with stm32f3xx-hal but not stm32f4xx-hal.  With stm32f4xx-hal and the stm32f401CxUx, the app hangs waiting inside write, inside write_read [here](https://github.com/stm32-rs/stm32f4xx-hal/blob/master/src/i2c.rs#L761).

I'll paste a few different backtraces below.  Would it be helpful for me to start looking at how to resolve this WriteRead implementation with the one that appears to be common to both f3xx-hal and h7xx-hal ? 

```
(gdb) bt
#0  0x0800224e in stm32f4::stm32f401::i2c1::sr1::<impl stm32f4::generic::R<u32,stm32f4::generic::Reg<u32,stm32f4::stm32f401::i2c1::_SR1>>>::sb (self=0x2000fd54)
    at /Users/todd/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32f4-0.9.0/src/stm32f401/i2c1/sr1.rs:864
#1  <stm32f4xx_hal::i2c::I2c<I2C,PINS> as embedded_hal::blocking::i2c::Write>::write (self=0x2000fef8, addr=118, bytes=...)
    at /Users/todd/.cargo/git/checkouts/stm32f4xx-hal-fe8350cc04cacf3f/62bd33c/src/i2c.rs:761
#2  0x080010c8 in <stm32f4xx_hal::i2c::I2c<I2C,PINS> as embedded_hal::blocking::i2c::WriteRead>::write_read (self=0x2000fef8, addr=118, bytes=..., buffer=...)
    at /Users/todd/.cargo/git/checkouts/stm32f4xx-hal-fe8350cc04cacf3f/62bd33c/src/i2c.rs:743
#3  0x080008e8 in bmp280_ehal::BMP280<I2C>::read_calibration (self=0x2000fef8)
    at /Users/todd/.cargo/registry/src/github.com-1ecc6299db9ec823/bmp280-ehal-0.0.3/src/lib.rs:69
```
		
```		
(gdb) bt
#0  0x0800226a in <stm32f4xx_hal::i2c::I2c<I2C,PINS> as embedded_hal::blocking::i2c::Write>::write (self=0x2000fef8, addr=118, bytes=...)
    at /Users/todd/.cargo/git/checkouts/stm32f4xx-hal-fe8350cc04cacf3f/62bd33c/src/i2c.rs:761
#1  0x080010c8 in <stm32f4xx_hal::i2c::I2c<I2C,PINS> as embedded_hal::blocking::i2c::WriteRead>::write_read (self=0x2000fef8, addr=118, bytes=..., buffer=...)
    at /Users/todd/.cargo/git/checkouts/stm32f4xx-hal-fe8350cc04cacf3f/62bd33c/src/i2c.rs:743
#2  0x080008e8 in bmp280_ehal::BMP280<I2C>::read_calibration (self=0x2000fef8)
    at /Users/todd/.cargo/registry/src/github.com-1ecc6299db9ec823/bmp280-ehal-0.0.3/src/lib.rs:69
#3  0x08000ac6 in bmp280_ehal::BMP280<I2C>::new (i2c=...) at /Users/todd/.cargo/registry/src/github.com-1ecc6299db9ec823/bmp280-ehal-0.0.3/src/lib.rs:59
```

```
(gdb) bt
#0  core::cell::UnsafeCell<T>::get (self=0x40005414) at /rustc/5e1a799842ba6ed4a57e91f7ab9435947482f7d8/src/libcore/cell.rs:1562
#1  0x0800223c in vcell::VolatileCell<T>::get (self=0x40005414) at /Users/todd/.cargo/registry/src/github.com-1ecc6299db9ec823/vcell-0.1.2/src/lib.rs:32
#2  stm32f4::generic::Reg<U,REG>::read (self=0x40005414) at /Users/todd/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32f4-0.9.0/src/generic.rs:51
#3  <stm32f4xx_hal::i2c::I2c<I2C,PINS> as embedded_hal::blocking::i2c::Write>::write (self=0x2000fef8, addr=118, bytes=...)
    at /Users/todd/.cargo/git/checkouts/stm32f4xx-hal-fe8350cc04cacf3f/62bd33c/src/i2c.rs:761
#4  0x080010c8 in <stm32f4xx_hal::i2c::I2c<I2C,PINS> as embedded_hal::blocking::i2c::WriteRead>::write_read (self=0x2000fef8, addr=118, bytes=..., buffer=...)
    at /Users/todd/.cargo/git/checkouts/stm32f4xx-hal-fe8350cc04cacf3f/62bd33c/src/i2c.rs:743
#5  0x080008e8 in bmp280_ehal::BMP280<I2C>::read_calibration (self=0x2000fef8)
    at /Users/todd/.cargo/registry/src/github.com-1ecc6299db9ec823/bmp280-ehal-0.0.3/src/lib.rs:69
#6  0x08000ac6 in bmp280_ehal::BMP280<I2C>::new (i2c=...) at /Users/todd/.cargo/registry/src/github.com-1ecc6299db9ec823/bmp280-ehal-0.0.3/src/lib.rs:59
```

tstellanova, 2020-03-01:
I after switching hardware I'm not able to reproduce this consistently, so I'm going to close the issue and chalk it up to a flaky hardware setup.

closed 2020-03-01 by tstellanova