stm32-rs/stm32f4xx-hal #141: Question: I2C blocking behaviour

Glaeqen, 2020-04-20:
Hi!
I have a question regarding I2C behaviour during failure scenario. 

Now, if I try to write some data to I2C interface and a target device is not there (disconnected or whatever), HAL implementation just hangs - [while loop waiting for a register to be cleared](https://github.com/stm32-rs/stm32f4xx-hal/blob/ddccb4fc48908f66f3848e686e87961a0df73a43/src/i2c.rs#L775).
```rust
// Wait until address was sent
while self.i2c.sr1.read().addr().bit_is_clear() {}
```

Apart from the fact that I don't know whether it is possible to implement it more sanely while satisfying `embedded_hal` API (I'm a rust newbie unfortunately 😞), I dare to say it is very much unacceptable behaviour to have. 

Unless I'm misunderstanding something, then sorry for a dumb question :/ 

Thank you for your response.

Glaeqen, 2020-05-04:
Is there anyone willing to reply? Thanks in advance :)

therealprof, 2020-05-04:
There's very little one can do about that unless the hardware provides a programmable watchdog. We could potentially provide something like a counter which the I2C implementation could use to check how much time has passed and eventually signal a timeout.

Glaeqen, 2020-05-04:
Hmm. I see. I guess that other implementations (like in CMSIS-Driver and vendor dependent ones) behave similarly. I wonder how does the mbed behave in such scenarios. Do you know anything about this by any chance? @therealprof 

Either way, watchdog sounds like a quick and reasonable solution to this issue in most cases.

therealprof, 2020-05-04:
Just to clarify, by watchdog I don't the MCU wide watchdog which restarts the device but a peripheral internal timer which is e.g. required for SMBus oeration which can detect and signal that soemthing is wrong on the bus.

I don't know what mbed does. There might be some operating systems which do all operations asynchronously in a thread and place a timeout on such activities but that's also a workaround at best if your hardware isn't behaving correctly.

Glaeqen, 2020-05-04:
Okey. Well, thank you for your reply! :)

closed 2020-05-04 by Glaeqen

Maldus512, 2021-04-22:
I'm following up on this to note that version 0.9.0 seems to fix the problem (correct me if I'm wrong) by leveraging on the `TIMEOUT` bit of the `I2C_SR1` register. I could verify that now querying a missing device results in a proper error instead of a blocked MCU.

therealprof, 2021-04-22:
[image]

That change was only done for completeness is a no-op (unless you have enabled SMBus mode). If you enable SMBus mode there're a couple of extra restrictions so it cannot be used regularly.

Maldus512, 2021-04-22:
That is correct, I've been to hasty - still, it seems that the problem of an unresponsive device has been fixed by 0.9.0, just not by the `TIMEOUT` bit. The loop that was originally discussed has become

```
 // Wait until address was sent
        while {
            // Check for any I2C errors. If a NACK occurs, the ADDR bit will never be set.
            let sr1 = self.check_and_clear_error_flags()?;

            // Wait for the address to be acknowledged
            sr1.addr().bit_is_clear()
        } {}

```

Among other things, `check_and_clear_error_flags()` checks for the `AF` (acknowledgement failure) bit, which was ignored in 0.8.3 while waiting for the first response. Thus it's not `Error::TIMEOUT` that gets returned, but `Error::NACK`, preventing the library code from blocking regardless.

Thanks for the clarification - and the amazing work.