stm32-rs/stm32f4xx-hal #166: Please add some DMA sample about Serial or other bus please.

XionWin, 2020-05-21:
This project is cool, and I use it on my stm32f407 device. but I can't find any sample and resource about serial port DMA and SPI DMA support. our cool project support DMA or not? Should I can get some sample about these? Thanks!

thalesfragoso, 2020-05-22:
DMA isn't yet supported, there is some work being done in https://github.com/stm32-rs/stm32f4xx-hal/pull/88, but I'm kinda busy with some other stuff at the moment, I also want to see how https://github.com/rust-embedded/embedonomicon/issues/64 goes before finishing it, since we will probably have new traits to work with.

XionWin, 2020-05-22:
Thanks for your reply, I got it, that's ok, I can find another way to achieve this requirement. good work, hope everything goes be fine.

closed 2020-05-22 by XionWin