stm32-rs/stm32f4xx-hal #176: serial example

pdgilbert, 2020-06-28:
I have been trying to write an example using `serial` that works on different MCUs/boards to help me better understand the advantages of `embedded-hal`. My current version uses a setup function to isolate the MCU specific setup part and separate it from the code that should be generic enough to work with different stm32??xx_hal HALs.

The code below compiles (and works) with `stm32f4xx_hal` but the need to specify details of the type returned by `serial1_setup()` seems unnecessarily messy.

```
//! Echo console input back to console + semihost output, char by char
//!
//! Connect the Tx pin pa9  to the Rx pin of usb-ttl converter
//! Connect the Rx pin pa10 to the Tx pin of usb-ttl converter
//! Set up the serial console (e.g. minicom) with the same settings used here.
//! (Using 9600bps, could be higher but needs serial console to be the same.)

#![deny(unsafe_code)]
#![no_main]
#![no_std]

#[cfg(debug_assertions)]
extern crate panic_semihosting;

#[cfg(not(debug_assertions))]
extern crate panic_halt;

use cortex_m_rt::entry;
use cortex_m_semihosting::hprintln;
use core::str::from_utf8;
use nb::block;

//use embedded_hal::serial;

#[cfg(feature = "stm32f4xx")] // eg Nucleo-64  stm32f411
use {stm32f4, stm32f4xx_hal::{prelude::*,  pac::Peripherals, serial::{config::Config, Serial }}};

#[entry]
fn main() -> ! {

    #[cfg(feature  = "stm32f4xx")]
    type SerialType = stm32f4xx_hal::serial::Serial<stm32f4::stm32f411::USART1,
    (stm32f4xx_hal::gpio::gpioa::PA9<stm32f4xx_hal::gpio::Alternate<stm32f4xx_hal::gpio::AF7>>,
     stm32f4xx_hal::gpio::gpioa::PA10<stm32f4xx_hal::gpio::Alternate<stm32f4xx_hal::gpio::AF7>>)>;

    #[cfg(feature = "stm32f4xx")]
    fn serial1_setup() -> SerialType {
        let cnfg = Config::default() .baudrate(9600.bps());
        let p = Peripherals::take().unwrap();
    	let rcc = p.RCC.constrain();
    	let clocks = rcc.cfgr.freeze();
    	let gpioa = p.GPIOA.split();
    	p.USART1.cr1.modify(|_,w| w.rxneie().set_bit());  //need RX interrupt? 
    	Serial::usart1(
    	    p.USART1,
    	    (gpioa.pa9.into_alternate_af7(),			      //rx pa9
	     gpioa.pa10.into_alternate_af7()),  		      //tx pa10
    	    cnfg,
    	    clocks,
    	    ).unwrap()
	}

    let mut txrx1 = serial1_setup(); 

    hprintln!("testwrite to console ...").unwrap();

    for byte in b"\r\nconsole connect check.\r\n" { block!(txrx1.write(*byte)).ok(); }

    hprintln!("test read and write by char. Please type into the console ...").unwrap();

    for byte in b"\r\nType (slowly) below:\r\n" { block!(txrx1.write(*byte)).ok(); }

    loop { // Read a byte and write
       let received = block!(txrx1.read()).unwrap();
       block!(txrx1.write(received)).ok();
       hprintln!("{}", from_utf8(&[received]).unwrap()).unwrap();
     }
}
```
With `stm32f1xx_hal`  I can 

`use stm32f1xx_hal::{prelude::*,  pac::Peripherals, serial::{Config, Serial, Tx1, Rx1 }}; 
`

and specify the signature of `serial1_setup()` as

`    fn serial1_setup(cnfg : Config) ->  (Tx1, Rx1) 
`

then split the result of `Serial::usart1()` before returning it, but `stm32f4xx_hal` does not seem to have any types or traits that allow me to do this cleanly.   (Possibly I should be using `embedded-hal` traits `Read` and `Write` but so far I have not figured out how to do that in a way that works.)

An example of how to initiate `serial` in a clean, proper, idiomatic way would be very much appreciated.

BTW. `stm32f4xx_hal` has `serial::config::Config` while `stm32f4xx_hal` and others I have looked at have `serial::Config`. It would be nice if they could be the same.

thalesfragoso, 2020-06-29:
The utility of `embedded-hal` is more to provide interoperability between binary and libraries (drivers), it isn't really expected to be able to run the same code on two different MCU's families.

therealprof, 2020-06-29:
@pdgilbert I have to agree with what @thalesfragoso said. However that doesn't mean we can't (or shouldn't) do better! Indeed types have been a common painpoint since we don't have GATs in the language yet, config gates can not be used everywhere and `impl Trait` also doesn't work in too many places.

> fn serial1_setup(cnfg : Config) -> (Tx1, Rx1)

`stm32f4xx-hal` has `Rx` and `Tx`, too but they're generic over the serial port type, e.g. `Tx<USART1>`.

Maybe in your particular case it would be an improvement to use the whole initialised serial port as is instead of splitting it into Tx/Rx which really only makes sense if those are used in independently which is rarely the case.

pdgilbert, 2020-06-29:
Well, I am new to this so I could be very confused. I do see the need to facilitate writing drivers with a common interface. It is not the driver talking to the hardware that I am wondering about, it is the other side of the driver, talking to code that wants to use it.  I'm not  sure what @thalesfragoso means by "interoperability between binaries and libraries (drivers)."  My application compiles into a binary that needs to interface with the drivers to use external peripherals, serial, and other on chip peripherals. So I do think of what I am doing as directly trying to use the "interoperability between binaries and libraries (drivers)."  Perhaps someone could explain the different interpretation that I am missing.

I would also appreciate understand what seems to be a different interpretation of the goals stated at
https://docs.rs/embedded-hal/1.0.0-alpha.1/embedded_hal/index.html. One of my earlier issues was being overly optimistic in trying to find a common API for hardware specific settings, which is clearly "out of scope". But the example above is trying to accomplish exactly what I see as the stated goals of `embedded_hal`. That is, in the setup function I am trying to do the hardware specific part, then pass the generic API back to my application code while erasing device specific details. I am just looking for a nice way to specify the return type from `Serial::usart1()`  or `Serial::usart1().split(()` so my code is less messy.

@therealprof,  I've been struggling with this for a few days.  `stm32f1xx_hal` has structs `Tx, Rx` and types `Tx1, Rx1` while  `stm32f4xx_hal `has structs `Tx, Rx`   and traits `PinTx, PinRx` and no types. It is the types in stm32f1xx_hal that allow using  `fn serial1_setup(cnfg : Config) -> (Tx1, Rx1)`. Possibly there is a good way to accomplish this using traits but, if so,  I have not figured it out. I did start out by trying to return the result of `Serial::usart1()`  but had no luck, while I have had some luck after splitting. I also think I should be using `Read` and `Write` from `embedded-hal` but I have not been able to make that work either.  I am really just looking for a nice clean coding example of how to make use of what I think are the stated goals of embedded_hal.

therealprof, 2020-06-29:
> Well, I am new to this so I could be very confused. I do see the need to facilitate writing drivers with a common interface. It is not the driver talking to the hardware that I am wondering about, [...]

`embedded-hal` provides the abstraction allowing to use abstracted functionality the same way and build drivers on top of it. The abstraction specifically does not (and cannot) cover the setup and the passing of peripherals following the abstraction. Which is why using different MCU families with the same codebase can be difficult or even impossible.

>  I am just looking for a nice way to specify the return type from Serial::usart1() or Serial::usart1().split(() so my code is less messy.

Different types are different types in Rust. It is not (easily) possible to gloss over that fact. There're some new possibilities like `impl Trait` (https://doc.rust-lang.org/stable/rust-by-example/trait/impl_trait.html) which can help getting over the strict type woes but those are actually nothing `embedded-hal` can help you with.

> @therealprof, I've been struggling with this for a few days. stm32f1xx_hal has structs Tx, Rx and types Tx1, Rx1 while stm32f4xx_hal has structs Tx, Rx and traits PinTx, PinRx and no types. It is the types in stm32f1xx_hal that allow using fn serial1_setup(cnfg : Config) -> (Tx1, Rx1)

I don't follow what you mean. `pub type Tx1 = Tx<USART1>;` means that type `Tx1` is an alias for the type `Tx<USART1>;`. The very same is possible for F4, too, though the type alias `Tx1` is a bit silly in my eyes. `stm32f1xx_hal::serial::Tx<USART1>` is a different type than `stm32f4xx_hal::serial::Tx<USART1>` and may have a different API so what you're asking is not easily possible.

It should be possible to write a generic function which takes any type implementing the function you're looking for. 

> I am really just looking for a nice clean coding example of how to make use of what I think are the stated goals of embedded_hal.

Maybe it is best if you look at how drivers can use arbitrary implementations of the `embedded-hal` traits. There're plenty of drivers on the https://github.com/rust-embedded/awesome-embedded-rust list, see e.g. https://github.com/dbrgn/rn2xx3-rs/blob/9f00e09385eee1fd9eaa01885dfa1f3e43a4a6d6/src/lib.rs#L366 for a driver using a serial port.

pdgilbert, 2020-06-30:
@therealprof writes:
> I don't follow what you mean. pub type Tx1 = Tx\<USART1\>; means that type Tx1 is an alias for the type Tx\<USART1\>;

Thank you. That was the hint I needed. `stm32f1xx_hal` exports `Tx1`. I did not understand it was an alias and did not understand the correct syntax for the underlying type. Using this I can specify the signature as 

`fn serial1_setup() ->  (Tx<USART1>, Rx<USART1>)`

and (after beating my head against the wall a bit to find the different places the different HALs put the pieces) I can use Tx, Rx, and USART1. This gives code that, other than the setup, is the same for all MCUs. It compiles and runs with `stm32f4xx_hal`,` stm32f4xx_hal`, and `stm32f4xx_hal`. @therealprof, I'm afraid it may take me awhile to digest some other parts of your response. On the surface, it seems to me you are saying that it may be difficult or impossible to do what I just did based on your clue. I think I must be missing the true meaning of what you said.

Here is the complete working example for reference. (It does not work on` stm32l1xx_hal` as that hal seems to be somewhat out of sync.)

```
//! Echo console input back to console + semihost output, char by char
//!
//! Connect the Tx pin pa9  to the Rx pin of usb-ttl converter
//! Connect the Rx pin pa10 to the Tx pin of usb-ttl converter
//! Set up the serial console (e.g. minicom) with the same settings used here.
//! (Using 9600bps, could be higher but needs serial console to be the same.)

#![deny(unsafe_code)]
#![no_main]
#![no_std]

#[cfg(debug_assertions)]
extern crate panic_semihosting;

#[cfg(not(debug_assertions))]
extern crate panic_halt;

use cortex_m_rt::entry;
//use core::fmt::Write;  // for writeln
use cortex_m_semihosting::hprintln;
use core::str::from_utf8;
use nb::block;

//use embedded_hal::serial;
//use embedded_hal::blocking::serial::{Write, Read};
//use embedded_hal::serial::{Write, Read};

#[cfg(feature = "stm32f1xx")]  //  eg blue pill stm32f103
use stm32f1xx_hal::{prelude::*,   
                    pac::Peripherals, 
                    serial::{Config, Serial, Tx, Rx},  
		    device::USART1 }; 

#[cfg(feature = "stm32f3xx")]  //  eg Discovery-stm32f303
use stm32f3xx_hal::{prelude::*, 
                    stm32::Peripherals,
                    serial::{ Serial, Tx, Rx},
		    stm32::USART1 };

#[cfg(feature = "stm32f4xx")] // eg Nucleo-64  stm32f411
use stm32f4xx_hal::{prelude::*,  
                    pac::Peripherals, 
                    serial::{config::Config, Serial, Tx, Rx},
		    pac::USART1 };

#[cfg(feature = "stm32l1xx") ] // eg  Discovery kit stm32l100 and Heltec lora_node STM32L151CCU6
use stm32l1xx_hal::{prelude::*, 
		    stm32::Peripherals, 
		    serial::{Config, Serial, Tx, Rx},
		    stm32::USART1 };

#[entry]
fn main() -> ! {

    // A simple abstraction for returning the result of Serial::usart1()  
    //  without  .split() inside serial1_setup()  still defeats me. 

    #[cfg(feature = "stm32f1xx")]
[... 63 lines cut ...]
    	    Config::default() .baudrate(9600.bps()),
    	    clocks,
    	    ).unwrap().split()
    	}

    // End of hal/MCU specific setup. Following should be generic code.

    let (mut tx1, mut rx1) = serial1_setup();

    hprintln!("testwrite to console ...").unwrap();

    for byte in b"\r\nconsole connect check.\r\n" { block!(tx1.write(*byte)).ok(); }

    hprintln!("test read and write by char. Please type into the console ...").unwrap();
    //writeln!(tx1, "\r\nPlease type (slowly) into the console below:\r\n").unwrap();
    for byte in b"\r\nType (slowly) below:\r\n" { block!(tx1.write(*byte)).ok(); }

    loop { // Read a byte and write
       let received = block!(rx1.read()).unwrap();
       block!(tx1.write(received)).ok();
       hprintln!("{}", from_utf8(&[received]).unwrap()).unwrap();
      }
}

```

closed 2020-06-30 by pdgilbert