stm32-rs/stm32f4xx-hal #180: Differences between this crate and stm32f1xx-hal

bschwind, 2020-07-16:
Hi there! I'm relatively new to the Rust embedded scene, but I've been able to get some simple firmware onto an `stm32f411`-based board and everything has worked great, I'm loving the way the type system guides me to using the correct pins for particular hardware peripherals.

I have both `stm32f411` boards and `stm32f103` boards. I started with the f411 but thought I'd try out some of my code on an f103. Immediately I noticed some differences between the crates and their APIs. As an example, the pwm code from both repos:

[f103 pwm example](https://github.com/stm32-rs/stm32f1xx-hal/blob/master/examples/pwm.rs)
[f411 pwm example](https://github.com/stm32-rs/stm32f4xx-hal/blob/master/examples/pwm.rs)

Lines like this have me a little confused:

```rust
// f103
let mut gpioa = p.GPIOA.split(&mut rcc.apb2);
...
let mut pwm = Timer::tim2(p.TIM2, &clocks, &mut rcc.apb1).pwm::<Tim2NoRemap, _, _, _>(
    pins,
    &mut afio.mapr,
    1.khz(),
);

```

```rust
// f411
let gpioa = dp.GPIOA.split();
...
let pwm = pwm::tim1(dp.TIM1, channels, clocks, 20u32.khz());
```

Why does the f411 code seem so much...simpler? Is its API design further along (or further behind)? Is the goal to get these two APIs closer than they currently are, or are they structured that way due to fundamental hardware differences?

I'd love to help out however I can, it's just hard for me to tell the state of these repos and where they're going. Thanks!

therealprof, 2020-07-16:
> Why does the f411 code seem so much...simpler? Is its API design further along (or further behind)? Is the goal to get these two APIs closer than they currently are, or are they structured that way due to fundamental hardware differences?

Often it's due to hardware differences, in these particular cases:
- your `split` example difference is due to philosophical reasons, the previous maintainer of the f1xx-hal wanted to have utmost correctness while we are a bit more relax, especially about peripheral configuration which is run exactly once at the very beginning of the application where races in a read-modify-write sequences are pretty much impossible
- the `pwm` difference is indeed due to a slightly more modern API design and hardware differences

In some cases it is possible to abstract over such differences and come up with a cross-family design, e.g. for USB, other times it's hard or even impossible.

bschwind, 2020-07-16:
Thanks for the explanation, that makes sense to me, especially regarding the more relaxed nature of the f4 crate and GPIO pin access. The differences are minimal enough to be able to easily work through them, and this explanation gives me a lot of context, thanks!

closed 2020-07-16 by bschwind

therealprof, 2020-07-16:
> relaxed nature of the f4 crate

Having said that, we could do better by using bitbanding. If someone wants to try that... ;)

thalesfragoso, 2020-07-16:
I don't personally like referring to this as relaxed, it's unsound, fortunately we have bit banding to help on some cases. However, I would expect (hope) that the gpio configurations will change.

therealprof, 2020-07-16:
> I don't personally like referring to this as relaxed

Well, I'm relaxed about this. 😅 I can't even think of a possibility where that could turn into a problem, to abuse this one would need to initialise a GPIO block within an interrupt handler and the main function (or another interrupt handler) "at the same time"  and the worst thing which could happen is that you're turning off a peripheral block which is completely safe from a program perspective.

I do think we should have support for proper initialisation sections where interrupts are guaranteed to be off and those would be the only places where you can get access to the PAC but that's a different discussion.

> However, I would expect (hope) that the gpio configurations will change.

Sure, if someone comes up with a good PR.

thalesfragoso, 2020-07-16:
> Well, I'm relaxed about this. I can't even think of a possibility where that could turn into a problem, to abuse this one would need to initialise a GPIO block within an interrupt handler and the [...]

There is also a data race in the case where you're changing gpio modes, you can end up with a mode completely different from the expected one which can cause all sort of behaviors.

therealprof, 2020-07-16:
> There is also a data race in the case where you're changing gpio modes, you can end up with a mode completely different from the expected one which can cause all sort of behaviors.

Yes, I agree, that is more of a problem, especially for people looking to do that actively all the time.

> Edit: Unfortunately, I don't think bit banding can help us in this case.

Indeed. More than 1 bit in a register => no bueno.