stm32-rs/stm32f4xx-hal #192: missing ADC2

pdgilbert, 2020-07-26:
Using recent git versions  I am getting
```
stm32::{ADC1, ADC2},
|              ^^^^
|              no `ADC2` in `stm32`
```
(nor is it in `pac`). According to release `doc.rs`, which may be out of sync with git,  I think it should be there, and `ADC1` is there.

Also, `Adc::adc2(...)` gives` function or associated item not found in `stm32f4xx_hal::adc::Adc<_>``

Is something missing or changed in the git version, or am I missing something?  The later is likely, I am still a newbie and my code is new and untested.

thalesfragoso, 2020-07-27:
What device feature are you using ? Probably not all chips have two ADCs.

pdgilbert, 2020-07-27:
That would explain it. I was trying with stm32f401 and stm32f411, which both only have 1 adc. This problem is fixed when building with stm32f405. Thanks.

closed 2020-07-27 by pdgilbert