stm32-rs/stm32f4xx-hal #198: UART8 interrupt not found

Maldus512, 2020-08-10:
I'm trying to setup a receiving interrupt for the UART8 peripheral. When declaring the interrupt handler however, the `UART8` variant is not found. When writing:

```
use stm32f4xx_hal::interrupt;
#[interrupt]
fn UART8() {
}
```

The compilation fails with the following error:

```
error[E0599]: no variant or associated item named `UART8` found for enum `stm32f4::stm32f413::Interrupt` in the current scope
  --> src/system/uart8.rs:42:4
   |
42 | fn UART8() {
   |    ^^^^^
   |    |
   |    variant or associated item not found in `stm32f4::stm32f413::Interrupt`
   |    help: there is a variant with a similar name: `UART5`
```

I'm compiling for the stm32f413 device, which is supposed to have an UART8 (I can read and write on it synchronously). I can write `USART8` instead of `UART8`, even if the 8th UART does not support synchronous mode; the compilation succeeds, but then the interrupt handler is never called (I am enabling it with `listen`).

I am confused over what the issue could be. Is there some reason why I shouldn't be able to use UART8 interrupts?

therealprof, 2020-08-10:
@Maldus512 I haven't checked the details here but it's likely you've just found a bug in the SVD files.

Please check with https://docs.rs/stm32f4/0.11.0/stm32f4/ and report at https://github.com/stm32-rs/stm32-rs if you find a discrepancy.

Maldus512, 2020-08-10:
@therealprof I suspected something along those lines, but I'm not experienced enough in Rust (macros are still out of my reach).
It does seem that the interrupt enum for stm32-rs is faulty (https://docs.rs/stm32f4/0.11.0/stm32f4/stm32f413/enum.Interrupt.html), mentioning USART8 instead of UART8. I'll relay the issue to them.

Thanks for the quick response!

closed 2020-08-10 by Maldus512

birkenfeld, 2020-08-10:
Just the name being wrong doesn't prevent the interrupt from working... Did you unmask the USART8 interrupt on the NVIC peripheral?

Maldus512, 2020-08-10:
Whoops... That is indeed the case. The name typo just distracted me. Thank you for the heads up!