stm32-rs/stm32f4xx-hal #202: i2c hangs when receiving NACK

ryan-summers, 2020-08-14:
The current I2C implementation will hang indefinitely whenever a NACK is received on the bus during a write request.

This occurs because the current implementation waits for the `SR1.addr` bit to be set before determining that the address transfer is complete. The RM notes that this bit does not assert when a NACK is received.

[image]

therealprof, 2020-08-14:
Interesting. Do you have a testcase for that?

ryan-summers, 2020-08-14:
On a project, we have a device with removable modules (8 of them). Each module is behind an I2C 1->8 mux.

In order to detect whether or not a module is installed, we try to read an I2C device on the bus. If the module is not properly installed (or the I2C device is in reset), I2C will hang indefinitely. I added a workaround such that both `addr` and `af` bits are checked and that seems to fix the problem.

ryan-summers, 2020-08-14:
It looks like we do check for a NACK, but _only_ in the data transfer phase, not at the address transfer phase.

An additional bug is that the NACK bit isn't cleared in `send_byte()`, so the NAK condition will persist from one transaction to the next.

ryan-summers, 2020-08-14:
A simple test case is to conduct an I2C write to a device that doesn't exist on the bus. The call will never finish.

ryan-summers, 2020-08-14:
I'll be happy to provide a patch for this by the way. Just wanted to formally open an issue to give it some exposure first. :)

therealprof, 2020-08-14:
I'd be happy to take a patch. Problem is more that this old I2C peripheral has so many cracks and crannies that it's hard to ensure that everything keeps working... 😅

commit 49b821bb (2020-08-15): Merge #203

203: Adding NACK detection and fault condition clearing r=therealprof a=ryan-summers

This PR fixes #202 by adding the following:
1. During all busy-wait I2C loops, error flags are continually checked.
2. All fault flags have been updated to manually clear the fault indication.

cc: @therealprof
PR #203 (2020-08-15): Adding NACK detection and fault condition clearing
This PR fixes #202 by adding the following:
1. During all busy-wait I2C loops, error flags are continually checked.
2. All fault flags have been updated to manually clear the fault indication.

cc: @therealprof
closed 2020-08-15 by bors[bot]