stm32-rs/stm32f4xx-hal #205: ADC initialization may panic

ryan-summers, 2020-08-20:
When constructing ADC3, I observed the initialization function panic. Upon debugging, it appears that the initializer is panicking during `adc::calibrate()` because `vref_samp` is set to zero.

Upon inspection of the STM32F4xx reference manual, I found this snippet:
[image]

It does not appear that it is valid to read the VREF_INT signal from any ADC other than ADC1.

therealprof, 2020-08-20:
Makes sense.

ryan-summers, 2020-08-20:
The result is that we don't have a calibrated VDDA for ADC2 and ADC3. Possible solutions:
* Allow user to manually specify VREF+ in the constructor
* Provide a function explicitly to acquire VREF+ calibration from the ADC1 peripheral and take that as a constructor argument for the other ADCs

commit 1dabe771 (2020-10-24): Merge #220

220: Updating ADC to support external reference specifications r=therealprof a=ryan-summers

This PR fixes #205 by updating the adc module to not allow access to VREF, VBAT, and the internal temperature sensor from ADC2 and ADC3, since these are only available to ADC1. To accomodate the loss of external reference calibration, the adc configuration has been updated to allow the user to specify the VDDA voltage.
PR #220 (2020-10-24): Updating ADC to support external reference specifications
This PR fixes #205 by updating the adc module to not allow access to VREF, VBAT, and the internal temperature sensor from ADC2 and ADC3, since these are only available to ADC1. To accomodate the loss of external reference calibration, the adc configuration has been updated to allow the user to specify the VDDA voltage.
closed 2020-10-24 by bors[bot]