stm32-rs/stm32f4xx-hal #226: SPI DMA transfer size support

albru123, 2020-11-05:
With current DMA implementation, it's impossible to configure DMA with 16-bit transfer size. It's also impossible to implement a wrapper type around SPI to define the u16 size manually, as you cannot implement the necessary `[DMASet](https://github.com/stm32-rs/stm32f4xx-hal/blob/385cb9707994d41a6199d1d139b49cfa4490a87d/src/dma/traits.rs#L313)` outside of the crate.

With the current structure, it would be necessary for [`PeriAddress`](https://github.com/stm32-rs/stm32f4xx-hal/blob/385cb9707994d41a6199d1d139b49cfa4490a87d/src/dma/traits.rs#L187) to accept transfer size as a generic type argument instead of having it as an associated type to enable multiple possible implementations for SPI (as you cannot implement `PeriAddress` twice with different `MemSize` when `MemSize` is an associated type). This change then propagates upwards up to the `[Transfer<>](https://github.com/stm32-rs/stm32f4xx-hal/blob/master/src/dma/mod.rs#L846)` type where another generic type argument defining the transfer size would be necessary.

This in fact makes perfect sense, as you cannot always assume the transfer size from the peripheral upon which you want to perform the DMA transfers.

```rust
type DmaTransfer = Transfer<Stream4<DMA1>, Channel0, SPI2, MemoryToPeripheral, &'static mut [u16; 64], u16 /* transfer size */>;
```

This is how the new `Transfer<>` type would probably look like when implemented naively like I proposed. There was also a proposition to hold the DMA transfer size associated with the SPI peripheral (as it usually depends on the SPI frame size), but I think it's not always desirable, as you could also use a [u8] source (something like BBQueue) to feed SPI with u16 frame size.

I think the transfer size parameter should be made a bit less confusing at least, by using something like an enum to define the desired size.

thalesfragoso, 2020-11-06:
Thanks for the report and investigation.

Having the `MemSize` as an associated type was probably an oversight on my part, I think it was because we already needed new types over some peripherals from day one (for timers with multiple possible registers), then I thought making a new type for configuration would make it clearer, but I didn't notice the problem with `DmaSet` and tuples.

So, I agree that having `MemSize` as a generic parameter is probably the best solution here.

> I think the transfer size parameter should be made a bit less confusing at least, by using something like an enum to define the desired size, if implemented like this.

Hmm, I think that can get a bit troublesome when matching `MemSize` with the word size of `WriteBuffer` from `embedded-dma`, but I will see if we can make that without having to make the internal code less readable.

thalesfragoso, 2020-11-06:
Oh, another thing, we might want to delay the 0.9.0 release because of this, hmm, let's try to solve this by the weekend.

albru123, 2020-11-06:
> Hmm, I think that can get a bit troublesome when matching MemSize with the word size of WriteBuffer from embedded-dma, but I will see if we can make that without having to make the internal code less readable.

Totally forgot about that. I mean, reasonable type parameter ordering would probably suffice, but it is a bit cluttered at the moment.

thalesfragoso, 2021-01-11:
I think #237 provides a solution to this. Closing here, feel free to re-open if this isn't the case.

closed 2021-01-11 by thalesfragoso

albru123, 2021-01-11:
Yup, it probably does, though I wasn't very successful with it yet 😅 

I'll let you know if something goes wrong.