stm32-rs/stm32f4xx-hal #236: Complementary PWM channels with dead time

kirill-havryliuk, 2020-11-22:
Hi, is it possible, to configure complementary PWM channels with dead time using stm32f4xx_hal?

kirill-havryliuk, 2020-11-22:
I, probably, can do it like:

```rust
    // TIM1_CH1  PE9
    // TIM1_CH1N PE8

    // TIM1_CH2  PE11
    // TIM1_CH2N PE10

    // TIM1_CH3  PE13
    // TIM1_CH3N PE12

    // Set complementary oututs mode as AF1
    gpioe.pe8 .into_alternate_af1();
    gpioe.pe10.into_alternate_af1();
    gpioe.pe12.into_alternate_af1();

    let channels = 
        ( gpioe.pe9 .into_alternate_af1()
        , gpioe.pe11.into_alternate_af1()
        , gpioe.pe13.into_alternate_af1()
        );

    let 
        ( mut ch1
        , mut ch2
        , mut ch3
        ) = pwm::tim1(device.TIM1, channels, clocks, 30000_u32.hz());

    unsafe {
        let tim1_regb = &(*(stm32::TIM1::ptr()));

        // Enable complementary outputs
        tim1_regb.ccer.modify(|_, w| w.cc1ne().set_bit());
        tim1_regb.ccer.modify(|_, w| w.cc2ne().set_bit());
        tim1_regb.ccer.modify(|_, w| w.cc3ne().set_bit());

        // Set dead time
        tim1_regb.bdtr.modify(|_, w| w.dtg().bits(10));

        // Center aligned
        tim1_regb.cr1.modify(|_, w| w.cms().center_aligned1());
    }

    let ch1_max_duty = ch1.get_max_duty();
    let ch2_max_duty = ch2.get_max_duty();
    let ch3_max_duty = ch3.get_max_duty();

    ch1.set_duty(ch1_max_duty / 2);
    ch2.set_duty(ch1_max_duty / 4);
    ch3.set_duty(ch1_max_duty / 8);

    ch1.enable();
    ch2.enable();
    ch3.enable();
```

I am very new to rust, so, maybe there exist better way to achieve that... Would appreciate any suggestions.

thalesfragoso, 2020-11-22:
Unfortunately, there isn't support for this on the HAL itself (yet).

About your example, you will need to check the manual and the HAL code to be sure your new configuration will play nice with our internal one.

kirill-havryliuk, 2020-11-24:
Thank you for you answer, @thalesfragoso. Provided example actually works. I haven't tested it good enough, but oscilloscope shows what I expect. 

"Complementary PWM outs + DT" is very essential feature, if you plan to drive bridge, for examples. 
It would be great to see it in HAL. 
But, anyway, thx, guys for your work!

thalesfragoso, 2020-11-25:
> It would be great to see it in HAL.

Yes, let's keep this open to track that.

newcomb-luke, 2021-09-30:
Has there been any work done to include complimentary PWM outputs at all in the HAL?

burrbull, 2024-06-01:
See https://github.com/stm32-rs/stm32f4xx-hal/pull/585/files

closed 2024-06-01 by burrbull