stm32-rs/stm32f4xx-hal #253: Request: Add API to expose TIMx peripheral from Timer<TIMx> struct

kesyog, 2021-01-20:
## Problem

Creating a `Timer` consumes the underlying `TIM` peripheral. For example: the `tim1` function consumes `TIM1`. However, sometimes I want to use features of the timer peripheral that aren't exposed by API's in the HAL crate.

In order to be able to use these non-exposed features, I think your options are:
1. Bypass the HAL crate entirely and access timer peripherals directly (e.g. via the peripheral access crate).
2. Add the needed API to the HAL crate.
3. Use unsafe code to steal another reference to the TIM2 peripheral registers via something like:
```rust
let mut hal_timer = hal::timer::Timer::tim2(TIM2, ...);
let pac_timer = unsafe { &mut *(hal::stm32::TIM2::ptr() as *mut hal::stm32::tim2::RegisterBlock) };
// do some things with pac_timer
```

## Proposal

Add a function to expose the underlying peripheral for all versions of `Timer` (and maybe something similar for other peripherals):

```rust
impl Timer<TIM2> {
    fn peripheral(&mut self) -> &mut TIM2 {
    }
}
```

This would allow someone to access the underlying registers without resorting to unsafe code.

Any reason why this would be a bad idea? Is there a different solution I missed above? This must be a commonly-encountered problem, but it doesn't seem like any of the stm32-rs HAL crates have this kind of API.

xoviat, 2021-01-21:
It's a bad idea because the timer needs exclusive control over the registers. Access to the registers is unsafe because it should be unsafe--because it could could the timer to stop working if they're modified. It would be safe to provide read-only access, but I haven't seen that done before.

kesyog, 2021-01-22:
Fair enough. I wouldn't say it'd be _unsafe_ for the current `Timer` implementation, but I could see how it'd go against the design patterns followed by the HAL trait.

I think I'm better off using unsafe to bypass the missing features in the short-term and adding proper support later.

closed 2021-01-22 by kesyog