stm32-rs/stm32f4xx-hal #268: Missing field

oqqA, 2021-02-20:
https://github.com/stm32-rs/stm32f4xx-hal/blob/5dcf835f06f0fce0d01bc13b6b82c4d6eeb53c10/examples/usb_serial.rs#L36

Missing field "hclk" in struct [stm32f4xx_hal::otg_fs::USB](https://docs.rs/stm32f4xx-hal/0.8.3/stm32f4xx_hal/otg_fs/struct.USB.html)

oqqA, 2021-02-20:
"usb_serial.rs" from branch "cargo-bloat" are correct

Please, replace example "usb_serial.rs" from branch "cargo-bloat" to branch "master"

thalesfragoso, 2021-02-20:
The examples match the current state of the code in the repository, not the current released crate version:
https://github.com/stm32-rs/stm32f4xx-hal/blob/5dcf835f06f0fce0d01bc13b6b82c4d6eeb53c10/src/otg_fs.rs#L17-L24

You can view the code and examples from released versions through the tags in the repo.

closed 2021-02-23 by oqqA