stm32-rs/stm32f4xx-hal #274: Spurious I2C bus errors occur

ryan-summers, 2021-03-17:
Due to errata in the STM32F4 I2C implementation, the I2C module may spuriously (and erroneously) indicate bus errors.

[image]

We should likely just throw out the BERR bit status instead of indicating incorrect errors, as advised by the errata.

The erratum is found in the following Errata sheets
Reference:
* [F42xx Errata](https://www.st.com/resource/en/errata_sheet/dm00068628-stm32f427437-and-stm32f429439-line-limitations-stmicroelectronics.pdf)
* [F40xx/F41xx Errata](https://www.st.com/resource/en/errata_sheet/dm00037591-stm32f405407xx-and-stm32f415417xx-device-limitations-stmicroelectronics.pdf)
* [STM32F46x/STM32F47x Errata](https://www.st.com/resource/en/errata_sheet/dm00220774-stm32f469xx-and-stm32f479xx-device-errata-stmicroelectronics.pdf)
* [F44x Errata](https://www.st.com/resource/en/errata_sheet/dm00155929-stm32f446xcxe-device-limitations-stmicroelectronics.pdf)

therealprof, 2021-03-17:
I haven't seen that behaviour in real life but it certainly makes sense to implement the suggested workarounds for the less lucky amongst us. 😅

ryan-summers, 2021-03-17:
We have been observing them spuriously on devices approximately once per day or so. The error is quite rare, but when you have a lot of bus transfers and have long running devices, it definitely props its head up. I'll add a fix for this soon.

commit c8b11b20 (2021-03-18): Merge #275

275: Errata workaround: Removing error return on I2C bus error detection r=therealprof a=ryan-summers

As outlined in #274, the I2C module may indicate spurious bus errors. This PR removes the `Error::BUS` variant from the I2C interface and updates the status flags check to ignore BERR bit assertions.

This fixes #274 

TODO:
- [x] I'm testing this update on our hardware to verify that it resolves our spurious reset issue
PR #275 (2021-03-18): Errata workaround: Removing error return on I2C bus error detection
As outlined in #274, the I2C module may indicate spurious bus errors. This PR removes the `Error::BUS` variant from the I2C interface and updates the status flags check to ignore BERR bit assertions.

This fixes #274 

TODO:
- [x] I'm testing this update on our hardware to verify that it resolves our spurious reset issue
closed 2021-03-18 by bors[bot]