stm32-rs/stm32f4xx-hal #285: bxcan/defmt version conflict

pdgilbert, 2021-04-02:
Could you please relax the version constraint on `bxcan` (#273) to ">=0.4" or  ">=0.4, <0.6" for reasons described at https://github.com/stm32-rs/stm32f7xx-hal/issues/114#issuecomment-812658708. Your CI on #273 looks clean, but you must not be testing with anything else that has a different specific `bxcan` or `defmt` version specification.

therealprof, 2021-04-02:
Indeed the tests only test against one specific version. Mind creating a PR for the suggested change?

pdgilbert, 2021-04-02:
Do you have a preference for  ">=0.4"  or  ">=0.4, <0.6". I'm a bit new to this and not sure what longer term maintenance issues will be for different choices.

therealprof, 2021-04-02:
I think I'd use the latter, if we need to adjust we can always adjust. Automatic breakage is bad and we don't know what the future will hold for us. ;)

pdgilbert, 2021-04-03:
Ok.  If I had been smarter I would have called my branch trying so I could run your CI first. I guess it gets run when I do the PR, so I won't mess with your ci.yml. I've done my own testing at https://github.com/pdgilbert/eg_stm_hal/actions/runs/712978531 and https://github.com/pdgilbert/LoRaGPS-rust/actions/runs/712984731. Everything works that is not broken for other reasons. I'll do the PR shortly.

pdgilbert, 2021-04-03:
Hmmm. Seems it failed the changelog check. Am I suppose to change that or do you?

therealprof, 2021-04-03:
You are supposed to do that. 😅

pdgilbert, 2021-04-03:
I'm not sure of the preferred way to indicat this change. The CHANGELOG.md `[Unreleased/Added]` section has
```
- Added support for canbus with the bxcan crate (0.5.0).
```
Should I edit that to make the version range clear, or add a new item line?

therealprof, 2021-04-03:
If it's in the `Unreleased` section you can just update the entry.

closed 2021-04-04 by pdgilbert