stm32-rs/stm32f4xx-hal #287: methods for opaque types

pdgilbert, 2021-04-03:
(I think this is a question not a bug.)

I have a setup function that implements on/off methods for an LED trait. This allows me to write code using `led.on()` and `led.off()` but switch which does` set_high()` and `set_low()`. This is useful for blackpill and nucleo-64 boards which both use `stm32f411` but are wired differently, so `set_high()` and `set_low()` have the opposite on/off results. Using the pin type this code works:

<details>
<summary>Click to expand</summary>

```
#![deny(unsafe_code)]
#![no_std]
#![no_main]

#[cfg(debug_assertions)]
use panic_semihosting as _;

#[cfg(not(debug_assertions))]
use panic_halt as _;

use cortex_m_rt::entry;

use embedded_hal::digital::v2::OutputPin;
use core::convert::Infallible;

pub trait LED {
    fn on(&mut self) -> ();
    fn off(&mut self) -> ();
}

use stm32f4xx_hal::{
    delay::Delay,
    gpio::{gpioc::PC13, Output, PushPull},
    //gpio::{gpioa::PA5, Output, PushPull,},
    pac::{CorePeripherals, Peripherals},
    prelude::*,
};

fn setup() -> (PC13<Output<PushPull>>, Delay) {
    let cp = CorePeripherals::take().unwrap();
    let p = Peripherals::take().unwrap();
    let rcc = p.RCC.constrain();
    let clocks = rcc
        .cfgr
        .hclk(48.mhz())
        .sysclk(48.mhz())
        .pclk1(24.mhz())
        .pclk2(24.mhz())
        .freeze();

    let gpioc = p.GPIOC.split();

    // Note that blackpill with stm32f411 and nucleo-64 with stm32f411 have onboard led wired
    // differently, so high/low setup is reversed (in addition to PA5 vs PC13).

    impl LED for PC13<Output<PushPull>> {
        fn on(&mut self) -> () {
            self.set_low().unwrap()
        }
        fn off(&mut self) -> () {
            self.set_high().unwrap()
        }
    }

    // return tuple  (led, delay)
    (
        gpioc.pc13.into_push_pull_output(), // led on pc13 with on/off
        Delay::new(cp.SYST, clocks),
    )
}

#[entry]
fn main() -> ! {
    let (mut led, mut delay) = setup();
    let on: u32 = 1000;
    let off: u32 = 3000;
    loop {
        let _r = led.on();
        delay.delay_ms(on);
        let _r = led.off();
        delay.delay_ms(off);
    }
}
```

</details>

I would like to change this to use implied types and traits but I cannot get the compiler to recognize the implementation. I think it should be something like

<details>
<summary>Click to expand</summary>

```
#![deny(unsafe_code)]
#![no_std]
#![no_main]

#[cfg(debug_assertions)]
use panic_semihosting as _;

#[cfg(not(debug_assertions))]
use panic_halt as _;

use cortex_m_rt::entry;

use embedded_hal::digital::v2::OutputPin;
use core::convert::Infallible;

pub trait LED {
    fn on(&mut self) -> ();
    fn off(&mut self) -> ();
}

use stm32f4xx_hal::{
    delay::Delay,
    gpio::{gpioc::PC13, Output, PushPull},
    //gpio::{gpioa::PA5, Output, PushPull,},
    pac::{CorePeripherals, Peripherals},
    prelude::*,
};

//fn setup() -> (PC13<Output<PushPull>>, Delay) {
fn setup() -> (impl OutputPin, Delay) {
    let cp = CorePeripherals::take().unwrap();
    let p = Peripherals::take().unwrap();
    let rcc = p.RCC.constrain();
    let clocks = rcc
        .cfgr
        .hclk(48.mhz())
        .sysclk(48.mhz())
        .pclk1(24.mhz())
        .pclk2(24.mhz())
        .freeze();

    let gpioc = p.GPIOC.split();

    // Note that blackpill with stm32f411 and nucleo-64 with stm32f411 have onboard led wired
    // differently, so high/low setup is reversed (in addition to PA5 vs PC13).

    //impl LED for PC13<Output<PushPull>> {
    impl LED for dyn OutputPin<Error = Infallible > {
        fn on(&mut self) -> () {
            self.set_low().unwrap()
        }
        fn off(&mut self) -> () {
            self.set_high().unwrap()
        }
    }

    // return tuple  (led, delay)
    (
        gpioc.pc13.into_push_pull_output(), // led on pc13 with on/off
        Delay::new(cp.SYST, clocks),
    )
}

#[entry]
fn main() -> ! {
    let (mut led, mut delay) = setup();
    let on: u32 = 1000;
    let off: u32 = 3000;
    loop {
        let _r = led.on();
        delay.delay_ms(on);
        let _r = led.off();
        delay.delay_ms(off);
    }
}
```

</details>

but I have tried several variation on this and always get

```
error[E0599]: no method named `on` found for opaque type `impl stm32f4xx_hal::prelude::_embedded_hal_digital_v2_OutputPin` in the current scope
  --> examples/zzblink.rs:72:22
   |
72 |         let _r = led.on();
   |                      ^^ method not found in `impl stm32f4xx_hal::prelude::_embedded_hal_digital_v2_OutputPin`
   |
   = help: items from traits can only be used if the trait is implemented and in scope
note: `LED` defines an item `on`, perhaps you need to implement it
  --> examples/zzblink.rs:16:1
   |
16 | pub trait LED {
   | ^^^^^^^^^^^^^

error[E0599]: no method named `off` found for opaque type `impl stm32f4xx_hal::prelude::_embedded_hal_digital_v2_OutputPin` in the current scope
...
```

Suggestions would be appreciate.

therealprof, 2021-04-03:
Well, the reason it says `LED` is not implemented is that you return `impl OutputPin`. If you returned `impl LED` it would work but I think you will need to `impl` for a specific type.

pdgilbert, 2021-04-03:
So when I change
```
impl LED for dyn OutputPin<Error = Infallible > {
```
to 
```
    impl LED for PC13<Output<PushPull>> {
```
and move it outside of the setup then I can change to
```
fn setup() -> (impl LED, Delay) {
```
and it builds. It's not clear that this improves the abstraction, I'll have to play a bit. Maybe there is no better way?

therealprof, 2021-04-03:
For me it works even inside `setup()`. I don't know of any better way but to me, it kinda looks like a nice abstraction because it puts all of the board specific configuration into `setup()`?

pdgilbert, 2021-04-03:
Ok, not sure what I did wrong, but it does work inside the `setup()`. My hope is to do all the board specific configuration in `setup()`.  I was over optimistically hoping I could impl LED without a specific type. Thanks for the explanation.

pdgilbert, 2021-04-03:
Just to complete the thread for anyone reading, this is the code that compiles:
```
#![deny(unsafe_code)]
#![no_std]
#![no_main]

#[cfg(debug_assertions)]
use panic_semihosting as _;

#[cfg(not(debug_assertions))]
use panic_halt as _;

use cortex_m_rt::entry;

use embedded_hal::digital::v2::OutputPin;

pub trait LED {
    fn on(&mut self) -> ();
    fn off(&mut self) -> ();
}

use stm32f4xx_hal::{
    delay::Delay,
    gpio::{gpioc::PC13, Output, PushPull},
    //gpio::{gpioa::PA5, Output, PushPull,},
    pac::{CorePeripherals, Peripherals},
    prelude::*,
};

fn setup() -> (impl LED, Delay) {
    let cp = CorePeripherals::take().unwrap();
    let p = Peripherals::take().unwrap();
    let rcc = p.RCC.constrain();
    let clocks = rcc
        .cfgr
        .hclk(48.mhz())
        .sysclk(48.mhz())
        .pclk1(24.mhz())
        .pclk2(24.mhz())
        .freeze();

    let gpioc = p.GPIOC.split();

    // Note that blackpill with stm32f411 and nucleo-64 with stm32f411 have onboard led wired
    // differently, so high/low setup is reversed (in addition to PA5 vs PC13).

    impl LED for PC13<Output<PushPull>> {
        fn on(&mut self) -> () {
            self.set_low().unwrap()
        }
        fn off(&mut self) -> () {
            self.set_high().unwrap()
        }
    }

    // return tuple  (led, delay)
    (
        gpioc.pc13.into_push_pull_output(), // led on pc13 with on/off
        Delay::new(cp.SYST, clocks),
    )
}

#[entry]
fn main() -> ! {
    let (mut led, mut delay) = setup();
    let on: u32 = 1000;
    let off: u32 = 3000;
    loop {
        let _r = led.on();
        delay.delay_ms(on);
        let _r = led.off();
        delay.delay_ms(off);
    }
}
```

closed 2021-04-03 by pdgilbert