stm32-rs/stm32f4xx-hal #296: closure form inside trait impl ?

pdgilbert, 2021-04-13:
(I think this is a question not a bug.)

With guidance from  @therealprof  in #287 I have a trait` LED` which gives me `.on()` and `.off()` bounds. I would like to add a bound (method?) `.blink()` which blinks the led on for a specified amount of time. This requires `delay`, which I think I need to capture from the environment of my `setup()` function, and which I also need to return from the `setup()` function to use in the main program. I am still trying to understand details of rust closures so I am not sure if I am attempting something that is not possible, or just have not got the proper incantation. The code below builds, but if I uncomment either of my attempts to pause inside `fn blink()` then I get `use the || { ... } closure form instead`.

What is the syntax for using the closure form inside the trait implementation?

<details>
<summary>Click to expand</summary>

```
#![deny(unsafe_code)]
#![no_std]
#![no_main]

#[cfg(debug_assertions)]
use panic_semihosting as _;

#[cfg(not(debug_assertions))]
use panic_halt as _;

use cortex_m_rt::entry;

use embedded_hal::digital::v2::OutputPin;

pub trait LED {
    fn on(&mut self) -> ();
    fn off(&mut self) -> ();
    fn blink(&mut self, on: u16) -> ();
}

use stm32f4xx_hal::{
    delay::Delay,
    gpio::{gpioc::PC13, Output, PushPull},
    //gpio::{gpioa::PA5, Output, PushPull,},
    pac::{CorePeripherals, Peripherals},
    prelude::*,
};

fn setup() -> (impl LED, Delay) {
    let cp = CorePeripherals::take().unwrap();
    let p = Peripherals::take().unwrap();
    let rcc = p.RCC.constrain();
    let clocks = rcc
        .cfgr
        .hclk(48.mhz())
        .sysclk(48.mhz())
        .pclk1(24.mhz())
        .pclk2(24.mhz())
        .freeze();

    let mut delay = Delay::new(cp.SYST, clocks);
    let mut pause = |t: u16| delay.delay_ms(t);

    // led
    let gpioc = p.GPIOC.split();
    let mut led = gpioc.pc13.into_push_pull_output(); // led on pc13

    // Note that blackpill with stm32f411 and nucleo-64 with stm32f411 have onboard led wired
    // differently, so this is reversed (in addition to PA5 vs PC13).

    impl LED for PC13<Output<PushPull>> {
        fn on(&mut self) -> () {
            self.set_low().unwrap()
        }
        fn off(&mut self) -> () {
            self.set_high().unwrap()
        }
        fn blink(&mut self, time: u16) -> () {
            self.on();
            //let _z = pause(time);
            //let _z = |time| pause(time);
            self.off()
        }
    }

    //led.blink(10000);
    led.on();
    let _z = pause(10000);
    led.off();

    // return tuple  (led, delay)
    (led, delay)
}
// End of hal/MCU specific setup. Following should be generic code.

#[entry]
fn main() -> ! {
    let (mut led, mut delay) = setup();

    let on: u32 = 1000;
    let off: u32 = 3000;

    loop {
        //led.blink(on);  would replace next 3 lines
        let _r = led.on();
        delay.delay_ms(on);
        let _r = led.off();
        delay.delay_ms(off);
    }
}
```
</details>

samcrow, 2021-04-15:
As far as I know, the language does not allow you to do this.

Your implementation `impl LED for PC13<Output<PushPull>>` captures the `pause` closure, which captures `delay`. However, this trait implementation has the same limitations as if you had written it at the top level (not inside any function). Specifically, it can't capture any variables from the surrounding function.

The error message is suggesting that you use a closure to define the `blink` function. That would let you capture `pause`, but you would not be able to implement a trait like that.

Unlike closures, trait implementations can be called from anywhere and the surrounding function may not even exist. That's why they can't capture variables.

Some alternatives:
* Make the delay an argument to the blink function
* Add a struct that contains the pin and a delay, and use that delay in the blink function

pdgilbert, 2021-04-15:
Thanks @samcrow for the explanation and the suggestions. I was beginning to think I had a "you cannot do that" kind of problem. But I'm still at the level of making simple silly mistakes so it is hard to be certain.

closed 2021-04-15 by pdgilbert

pdgilbert, 2021-04-15:
(Just to complete the thread for anyone reading.) This works when `delay` is an argument to a `blink` function defined in `main` :
<details>
<summary>Click to expand</summary>

```
#![deny(unsafe_code)]
#![no_std]
#![no_main]

#[cfg(debug_assertions)]
use panic_semihosting as _;

#[cfg(not(debug_assertions))]
use panic_halt as _;

use cortex_m_rt::entry;

use embedded_hal::digital::v2::OutputPin;

pub trait LED {
    fn on(&mut self) -> ();
    fn off(&mut self) -> ();
}

use stm32f4xx_hal::{
    delay::Delay,
    gpio::{gpioc::PC13, Output, PushPull},
    //gpio::{gpioa::PA5, Output, PushPull,},
    pac::{CorePeripherals, Peripherals},
    prelude::*,
};

fn setup() -> (impl LED, Delay) {
    let cp = CorePeripherals::take().unwrap();
    let p = Peripherals::take().unwrap();
    let rcc = p.RCC.constrain();
    let clocks = rcc
        .cfgr
        .hclk(48.mhz())
        .sysclk(48.mhz())
        .pclk1(24.mhz())
        .pclk2(24.mhz())
        .freeze();

    let delay = Delay::new(cp.SYST, clocks);

    // led
    let gpioc = p.GPIOC.split();
    let led = gpioc.pc13.into_push_pull_output(); // led on pc13

    // Note that blackpill with stm32f411 and nucleo-64 with stm32f411 have onboard led wired
    // differently, so this is reversed (in addition to PA5 vs PC13).

    impl LED for PC13<Output<PushPull>> {
        fn on(&mut self) -> () {
            self.set_low().unwrap()
        }
        fn off(&mut self) -> () {
            self.set_high().unwrap()
        }
    }

    // return tuple  (led, delay)
    (led, delay)
}
// End of hal/MCU specific setup. Following should be generic code.

#[entry]
fn main() -> ! {
    let (mut led, mut delay) = setup();

    fn blink(led : &mut impl LED, time: u16, delay : &mut Delay ) -> () {
        led.on();
        delay.delay_ms(time);
        led.off()
    }

    let on:  u16 = 1000;
    let off: u16 = 3000;

    loop {
        blink(&mut led, on, &mut delay);
        delay.delay_ms(off);
    }
}
```
</details>

And also works when the `blink` function is an `LED` trait bound provided `delay` is passed as an argument:
<details>
<summary>Click to expand</summary>

```
#![deny(unsafe_code)]
#![no_std]
#![no_main]

#[cfg(debug_assertions)]
use panic_semihosting as _;

#[cfg(not(debug_assertions))]
use panic_halt as _;

use cortex_m_rt::entry;

use embedded_hal::digital::v2::OutputPin;

pub trait LED {
    fn on(&mut self) -> ();
    fn off(&mut self) -> ();
    fn blink(&mut self, time: u16, delay : &mut Delay ) -> ();
}

use stm32f4xx_hal::{
    delay::Delay,
    gpio::{gpioc::PC13, Output, PushPull},
    //gpio::{gpioa::PA5, Output, PushPull,},
    pac::{CorePeripherals, Peripherals},
    prelude::*,
};

fn setup() -> (impl LED, Delay) {
    let cp = CorePeripherals::take().unwrap();
    let p = Peripherals::take().unwrap();
    let rcc = p.RCC.constrain();
    let clocks = rcc
        .cfgr
        .hclk(48.mhz())
        .sysclk(48.mhz())
        .pclk1(24.mhz())
        .pclk2(24.mhz())
        .freeze();

    let delay = Delay::new(cp.SYST, clocks);

    // led
    let gpioc = p.GPIOC.split();
    let led = gpioc.pc13.into_push_pull_output(); // led on pc13

    // Note that blackpill with stm32f411 and nucleo-64 with stm32f411 have onboard led wired
    // differently, so this is reversed (in addition to PA5 vs PC13).

    impl LED for PC13<Output<PushPull>> {
        fn on(&mut self) -> () {
            self.set_low().unwrap()
        }
        fn off(&mut self) -> () {
            self.set_high().unwrap()
        }
        fn blink(&mut self, time: u16, delay : &mut Delay ) -> () {
            self.on();
            delay.delay_ms(time);
            self.off()
        }
    }

    // return tuple  (led, delay)
    (led, delay)
}
// End of hal/MCU specific setup. Following should be generic code.

#[entry]
fn main() -> ! {
    let (mut led, mut delay) = setup();

    let on:  u16 = 1000;
    let off: u16 = 3000;

    loop {
        led.blink(on, &mut delay); 
        delay.delay_ms(off);
    }
}
```
</details>