stm32-rs/stm32f4xx-hal #326: USB Serial example doesn't work on stm32f405

mythmon, 2021-07-01:
Hi! I'm trying to run the [usb_serial example from the this repo](https://github.com/stm32-rs/stm32f4xx-hal/blob/master/examples/usb_serial.rs). After building the example and flashing it, when I plug the microcontroller back in to my computer I get an error from Windows that says the device was not recognized, and is malfunctioning. In Linux, it says "device not responding to setup address" and "device not accepting address 4, error -71".

To compile the example for stm32f405 I changed the features listed in Cargo.toml to target the new device.

I'm using [this Adafruit Feather board](https://www.adafruit.com/product/4382). I have been able to control the onboard LED using this HAL crate already.

thalesfragoso, 2021-07-01:
What optimizations options are you using ? I suggest compiling with `--release` with something like:
```toml
[profile.release]
opt-level = "s"
lto = true
debug = true
codegen-units = 1
```

Also make sure to change this line to your correct HSE value:
https://github.com/stm32-rs/stm32f4xx-hal/blob/1f5750313511e1b00f648b8fc09cdfb6562b2d47/examples/usb_serial.rs#L23

mythmon, 2021-07-02:
I had used both opt-level `"s"` and `3`, to the same effect. However, updating the HSA frequency (12 MHz for this board) fixed the problem. Thanks for the help!

closed 2021-07-02 by mythmon

gretchen6, 2021-12-21:
@mythmon could you please elaborate on your solution?
I am also using the stm32f405 board and got the led blinking.
I tried both 12 MHz and 25MHz for the hse frequency, however without any luck.
According to the [data sheet](https://www.st.com/resource/en/datasheet/stm32f405rg.pdf) on page 93 it should be 25 MHz.

I am testing on a linux machine and the error is that `/dev/ttyACM0` does not show up.

mythmon, 2021-12-21:
I think it was pure luck that I was able to figure it out, to be honest. I've since stopped working on this, and I don't remember how I got it working in the first place, sorry.

gretchen6, 2021-12-21:
Oh well. Thanks for the answer though!

adamgreig, 2021-12-21:
Make sure the HSE is whatever is actually on your board: the datasheet is just saying what current consumption is when you use a 25MHz HSE, but it may be anything from 1MHz up to 50MHz. What board are you using?

gretchen6, 2021-12-21:
@adamgreig I am unsure where I can properly figure out which HSE is correct. I found a post mentioning 16MHz but that doesn't seem to have an effect as well. When I print something via usb cable using Serial while programming in the Arduino IDE it works just fine, so I think my setup isn't completely broken.

I have the 'Adafruit Feather STM32F405' board

adamgreig, 2021-12-22:
Wow, they don't make it easy to find out! In the end I found the schematic image for the "Adafruit STM32F405 Feather Express" which has a 12MHz HSE fitted, so that's the number to use. If it's still not working with that I guess something else is wrong but I'm not sure what...

gretchen6, 2021-12-22:
Thanks a lot for figuring the number out!
While I'm quite sure I tried that value out before, I'm much more glad that it works now :)