stm32-rs/stm32f4xx-hal #362: ADC voltage conversion might be incorrect

HarryMakes, 2021-09-03:
Hello, I would like to know if there is an error for ADC digital-to-voltage conversion. Currently, it uses `voltage = ADC_code * Vref / max_sample`, where `max_sample` is always `1 << bit_resolution - 1` instead of just `1 << bit_resolution`.

https://github.com/stm32-rs/stm32f4xx-hal/blob/979ff1555f79e3e268a600caa5c43b8d3de95217/src/adc.rs#L796-L801

This would be the contrary of most ADCs, where the divisor is **2^bits**. There's also a comment on StackExchange claiming that this should be the standard ([link](https://stackoverflow.com/a/7895054)).

It makes more sense to me that (2^bits-1) is indeed incorrect, but I'm also surprised that the STM32 official HAL seems to adopt the same approach as this Rust HAL. Relevant code here: [link](https://github.com/STMicroelectronics/STM32CubeF4/blob/4aba24d78fef03d797a82b258f37dbc84728bbb5/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_ll_adc.h#L1603-L1604).

It'd be great to hear from others. Thanks!

adamgreig, 2021-09-03:
Amusingly all the other answers on that StackExchange question propose using 4095, mostly justifying it because if you pick 4096 you'll never compute the voltage as 5.000V, even if 5.000V is being applied. I think "check the datasheet" is the only good answer on that question, as ADCs can all differ.

Indeed the STM32F405 datasheet delivers (and the other STM32 datasheets are the same, in fact this figure is used in all the STM32 datasheets I checked even for higher-precision ADCs :thinking:), specifying `1LSB(ideal)=Vref+/4096` in this somewhat confusing chart:

[image]

The upper staircase labelled (2) is the ideal transfer function, i.e. voltages along the x-axis would map to the ADC code on the y-axis if the ADC had zero gain, offset, and linearity error.

Here the ADC code `n` represents an input voltage between `Vref*(n-0.5)/4096` and `Vref*(n+0.5)/4096`, in other words code 0 means the voltage was ideally between +-403nV (at vref=3.3v), and code 4095 means the voltage was ideally between 3.2988V and 3.2996V. There's no code for voltage 3.3V itself. If we return `vref*n/4096` we output the midpoint of these bins, in other words code 0 maps to 0.0000V and code 4095 maps to 3.2992V. If we use `vref*n/4095` instead, we effectively add 1LSb of gain error on top of the typical value of 1.5LSb for this STM32F405.

As you mentioned this is the same or similar for most ADCs as well. For example this random AD ADC is somewhat more clearly specified:

[image]

The maximum ADC code here corresponds to an input voltage equal to or exceeding FS - 3/2 LSB just like the STM32. Its scale factor is likewise `1LSb = V/(2^n)`.

Either way the scale in use should be clearly documented so that users know the largest output won't be the reference, in case they're using it for something where they expect that to be the case or expect identical behaviour to the (buggy?) ST HALs.

HarryMakes, 2021-09-06:
@adamgreig Thanks for the input! I'm developing on an STM32F407 and its datasheet also shows a similar figure for ADC accuracy. As shown, both the actual and ideal transfer curve have the same slope. This means each increment in the ADC code still corresponds to `VREF+/(2^n)` increment in the voltage - correct?

I think, "ideal vs actual" should concern more the offset/gain errors (first/last transition), rather than the voltage that 1 LSB correspond to.

Before reading the datasheet, I also came across [this ST manual](https://www.st.com/resource/en/application_note/cd00211314-how-to-get-the-best-adc-accuracy-in-stm32-microcontrollers-stmicroelectronics.pdf). It went in detail (in section 3.1.2) how the "last code transition" contributes to a negative gain error. In section **3.1.1** it specifically says 1 LSB equals `VREF+/4096` ("smallest detectable change") as well, so I'm now very inclined to think that the HAL implementations are "wrong" - or rather, inaccurate.

In my opinion, although 4095 vs 4096 as the divisor probably should only account for a small portion of measurement error in practice, the HAL implementations are still conceptually conflicting. I don't know if the HALs (both stm32-rs and ST's C lib) should be fixed and/or documentation should be improved to address this issue.

adamgreig, 2021-09-06:
> As shown, both the actual and ideal transfer curve have the same slope. This means each increment in the ADC code still corresponds to VREF+/(2^n) increment in the voltage - correct?

That's right. Be careful that in the F407 datasheet it incorrectly labels the notes, 1 should be 'example of actual', 2 should be 'ideal', 3 should be 'end point correlation line', all fixed in later datasheets with the same figure. The two dashed diagonal lines are the endpoint correlations, rather than the transfer curves. The 'example of actual' is one of many possible actual transfer curves for a specific device depending on its errors. Nevertheless 1/4096 is still the right slope.

> In my opinion, although 4095 vs 4096 as the divisor probably should only account for a small portion of measurement error in practice, the HAL implementations are still conceptually conflicting. I [...]

Yea, I agree. It's probably insignificant in the scheme of things (especially since anyone doing precision DSP is probably not converting to a floating point voltage like this) but it does look wrong.

gernoteger, 2021-12-07:
When I was wondering why my ADC readings systematically were differing from measurements I came across this issue. Fixing max_sample to 1<<n instead of 1<<n -1 brought my readings reasonably close to the measurements.
Are there any objections to actually fix the code?

commit d9a9cd25 (2021-12-11): fix #362: ADC voltage conversion might be incorrect
commit 9424a372 (2021-12-11): Merge #397

397: fix #362: ADC voltage conversion might be incorrect r=burrbull a=gernoteger

since the discussion in #362 seems to come to a conclusion, and my personal measurements supported that, I think it's time to actually fix it.
This will change max_sample from 1<<n-1 to 1<<n,
After this PR ADC readings will change, but will be more accurate than before. Users might be confused since they won't get a reading of 3300mV, but this is correct behaviour.
closed 2021-12-11 by bors[bot]