stm32-rs/stm32f4xx-hal #369: Cannot resolve cortex-m-rt dependency in 0.10.0

TWALL9, 2021-10-06:
Found earlier today while updating my dependencies, but confirmed while trying to build the examples in this repo.  When using interrupts via the `rt` feature, the dependencies for `cortex-m-rt` in `stm32f4xx_hal` (^0,7) conflict with the version of `cortex-m-rt` required by `stm32f4` (^0.6.10, 0.6.15).  This can be avoided if `cortex-m-rt = "0.7"` is used while `stm32f4xx_hal = "0.10.0"` explicitly does not use the `rt` feature.  However, this removes the ability to use hardware peripheral interrupts.  In version 0.9.0, `rt` works fine

I have a feeling that this is also the core problem behind [issue 366](https://github.com/stm32-rs/stm32f4xx-hal/issues/366)

Output while building the blinky-timer-irq example

```shell
cargo build --example blinky-timer-irq 
    Updating crates.io index
error: failed to select a version for `cortex-m-rt`.
    ... required by package `stm32f4 v0.13.0`
    ... which satisfies dependency `stm32f4 = "^0.13"` of package `stm32f4xx-hal v0.10.0 (/home/tom/emb-rust/stm32f4xx-hal)`
versions that meet the requirements `^0.6.10` are: 0.6.15

the package `cortex-m-rt` links to the native library `cortex-m-rt`, but it conflicts with a previous package which links to `cortex-m-rt` as well:
package `cortex-m-rt v0.7.0`
    ... which satisfies dependency `cortex-m-rt = "^0.7"` of package `stm32f4xx-hal v0.10.0 (/home/tom/emb-rust/stm32f4xx-hal)`
Only one package in the dependency graph may specify the same links value. This helps ensure that only one copy of a native library is linked in the final binary. Try to adjust your dependencies so that only one package uses the links ='cortex-m-rt' value. For more information, see https://doc.rust-lang.org/cargo/reference/resolver.html#links.

failed to select a version for `cortex-m-rt` which could resolve this conflict

```

TWALL9, 2021-10-06:
Ah, silly me I see that `stm32f4 = "0.14.0"` is in Cargo.toml now.  Still though, anyone updating to 0.10.0 is in for a shock.  0.10.1 when?

adamgreig, 2021-10-06:
Ah, yep, it looks like stm32f4xx-hal 0.10 shouldn't have been released with a cortex-m-rt 0.7 and stm32f4 0.13 dependency side-by-side, since they can't co-exist. Maybe it would make sense to yank and release 0.10.0 with stm32f4 0.14.

burrbull, 2021-10-06:
What if release 0.10.1 with ">=0.6.15, <0.8" ?

dbrgn, 2021-11-07:
> What if release 0.10.1 with ">=0.6.15, <0.8" ?

This only helps in the case where the project is still using cortex-m-rt 0.6, but not if it's using cortex-m-rt 0.7, right? The HAL still depends on stm32f4 0.13, which in turn depends on cortex-m-rt 0.6.

If I try to work around this by removing the `"rt"` feature, compilation fails with:

> error[E0433]: failed to resolve: could not find `interrupt` in `you_must_enable_the_rt_feature_for_the_pac_in_your_cargo_toml`

If I downgrade to cortex-m-rt 0.6 it seems to work.

KoffeinFlummi, 2021-11-09:
> It would be nice if a 0.11 could be released, which depends on cortex-m-rt 0.7 and on stm32f4 0.14.

I would agree. I just tried updating my dependencies, and using the exact `[dependencies]` from the README, I still encountered this.