stm32-rs/stm32f4xx-hal #391: UART doesn't seem to send any character.

MOWA2, 2021-11-22:
Hi,

I'm currently testing Rust in embedded developement. It has been great so far but i'm facing an issue with the UART.

I have a F411RE board. I compiled and ran the example code for UART but I cannot receive anything on the other end.

Unfortunately, I do not have any oscilloscope to figure if any signal is coming from the TX pin.

Do someone has the same problem ?

Cheers.

burrbull, 2021-11-24:
I can't reproduce on F411CE. Not sure the problem in UART.
Did you compile in `--release`?
Have you tried `serial` example from `examples` directory?
Have you tried blinky example to be sure you flash tool work as expected? What about `memory.x`?
...

See this repo for quick start on F411RE: https://github.com/kalkyl/f411-rtic/

MOWA2, 2021-11-24:
Hi,

Thank you for your quick response ! 😄

I tried the `—release` compile option, but did not change anything.

`Memory.x` is good, and my setup seems ok. I can run a blinky program without any issues.

Thank you for your link, I’ll give it a shot as soon as I come back home ! :)

Br;

MOWA2, 2021-11-24:
Hi again.

Can't figure it out.. Program is running but USART isn't sending anything.

Do you see something wrong ?

```
/*
 * Author : CJ
 * File : main.rs --> Entrypoint
 **/

/* Embedded RUST directives */
#![no_std]
#![no_main]

/* Panic handler in case of errors */
use panic_halt as _;

/* Cortex M main macros/startup codes */
use cortex_m::Peripherals as _core_periph;
use cortex_m_rt::entry;
use cortex_m_semihosting::hprintln;
use stm32f4xx_hal::block;
//use cortex_m_rt::interrupt;

/* STM32F4xx HAL */
use stm32f4xx_hal::delay::Delay;
use stm32f4xx_hal::serial::{config, Serial};
use stm32f4xx_hal::prelude::*;
use stm32f4xx_hal::pac::Peripherals as _device_periph;

use core::fmt::Write; // for pretty formatting of the serial output

mod peripherals_initialisation;
mod log;

#[entry]
fn main() -> ! {

    /* Get access to device and core peripherals */
    let mut core_peripherals = _core_periph::take().unwrap();
    let mut device_peripherals = _device_periph::take().unwrap();

    /* Get access to RCC and GPIO */
    let rcc = device_peripherals.RCC.constrain();
    let gpioa = device_peripherals.GPIOA.split();
    let gpiob = device_peripherals.GPIOB.split();

    /* Set up sysclk to max (100mhz) and freeze it */
    let clocks = rcc.cfgr.sysclk(100.mhz()).freeze();

    /* Set up systick delay */
    let mut delay = Delay::new(core_peripherals.SYST, &clocks);

    /* Set up embedded LED pin */
    let mut embedded_led = gpioa.pa5.into_push_pull_output();

    let mut tx_pin = gpioa.pa2.into_alternate();
    //let mut rx_pin = gpioa.pa3.into_alternate();

    let usart_config = config::Config::default().baudrate(9600.bps());

    let mut tx = Serial::tx(device_peripherals.USART2, tx_pin, usart_config, clocks).unwrap();

    //block!(tx.write(b'X')).ok();
    writeln!(tx, "value: {:02}\r\n", 4).unwrap();
    //block!(writeln!(tx, "value: {:02}\r\n", 4).unwrap()).ok();

    let hello: &str = "Hello, I'm a STM32F4xx!\r\n";
    loop {
        /* Light show */
        tx.write_str(hello).unwrap();

        embedded_led.set_high();
        delay.delay_ms(1_000_u16);
        embedded_led.set_low();
        delay.delay_ms(1_000_u16);
    }
}
```

memory.x

```
/* memory.x - Linker script for the STM32F411RE */

MEMORY
{
    /* Flash memory begins at 0x80000000 and has a size of 512kB*/
    FLASH : ORIGIN = 0x08000000, LENGTH = 512K
    /* RAM begins at 0x20000000 and has a size of 128kB*/
    RAM : ORIGIN = 0x20000000, LENGTH = 128K
}

_stack_start = ORIGIN(RAM) + LENGTH(RAM);
```

Cargo.toml

```
[profile.release]
opt-level = 'z'                 # Turn on maximum optimizations.
lto = true                      # Link-time-optimizations for further size reduction

[dependencies]
cortex-m = "0.7.3"             # Access to the generic ARM peripherals
cortex-m-rt = "0.6.15"         # Startup code for the ARM Core
panic-halt = "0.2.0"           # Panic handler
cortex-m-semihosting = "0.3.7"

# Access to the STM32F411 HAL.
[dependencies.stm32f4xx-hal]
# STM32F411RE contains a 512kB flash variant which is called "medium density"
features = ["stm32f411", "rt"]
version = "0.10.1"

```

Thanks.

burrbull, 2021-11-25:
Don't see any issues in software. Are you sure receiver has same baudrate/config?
Do you connect `Device1 TX <-> RX Device2` ?

MOWA2, 2021-11-25:
Yep, I tested it with ESP32 which has 3,3v UART and with Arduino which has 5V UART but CMOS 2,7V trigger on RX pin.

JohnnnyJohnnny, 2022-01-06:
I think your pins aren't connected to the pinheader. 
The default configuration on the nucleo board is that D0/D1 (PA2/PA3) are exclusively connected to the ST-Link. You have to change two 0 Ohm resistors on the back to use them. 

The example was ment to use with a PC. When you connect your Nucleo board to the PC, a virtual com port also registred aside from the debugger. So you probably see your "hello" with Putty or something similar.

Try Uart1 with PA9 and PA10 for your problem.

alexxy, 2022-01-19:
I have same problem on black pill with stm32f411ce

MOWA2, 2023-03-19:
> I think your pins aren't connected to the pinheader. The default configuration on the nucleo board is that D0/D1 (PA2/PA3) are exclusively connected to the ST-Link. You have to change two 0 Ohm [...]

Changing UART just fixed the problem, thank you.

closed 2023-03-19 by MOWA2