stm32-rs/stm32f4xx-hal #410: Quadrature Encoder Interface only supports rising edge polarity

coryalder, 2022-01-05:
In the timer encoder mode, we have a hard-coded configuration of the polarity the encoder input: https://github.com/stm32-rs/stm32f4xx-hal/blob/d96137099d47112bf891d72df1c066409afacb4e/src/qei.rs#L79

and on the following lines cc1p and cc2p are cleared, setting the polarity of the QEI. Since the QEI supports two polarities, this interface should probably have a way to change the polarity - either as an input parameter to `setup_qei` e.g. `setup_qei(Polarity::FallingEdge)` or as a function we can call on the resulting Qei type - something like `set_polarity(Polarity::FallingEdge)`.

As a workaround in my code, I'm currently running this immediately after calling `setup_qei`:

    unsafe {
        (*TIM1::ptr()).ccer.write(|w|{ w.cc1p().set_bit().cc2p().set_bit() });
    }

But it would be grreat to have a safe interface for this.

burrbull, 2022-01-05:
Good idea.
Can this be changed immediately or it needs timer restart or something?

coryalder, 2022-01-05:
I'm not restarting the timer or anything and it appears to be working. I call `Qei::new` then my workaround immediately after.