stm32-rs/stm32f4xx-hal #411: USB Data interrupt

zgdump, 2022-01-07:
Hello. How do I get an interrupt when data arrives via USB? I asked in another repository, I was shown an [old example](https://github.com/stm32-rs/stm32-usbd-examples/blob/master/example-stm32f103c8/examples/serial_rtfm.rs) for f103. But it uses USB_HP_CAN_TX and USB_LP_CAN_RX0, which are not present for f411 in this HAL. What should I do?

TeXitoi, 2022-01-07:
https://github.com/TeXitoi/keyberon-f4/blob/master/src/main.rs#L120-L128

zgdump, 2022-01-10:
@TeXitoi thank you! It really works. But it confuses me that the `rprint` functions do not work in the interrupt handler, but start working in any other functions called from the same interrupt. Magic :)

closed 2022-01-10 by zgdump

Slamy, 2022-08-27:
This is the issue I was looking for. I'm pretty new to embedded rust and tried to change the usb_serial.rs example to utilize the OTG_FS interrupt. Today is now the third day I'm trying to do this and I keep failing even after reading a lot about the life time checker. Curious ones might be able to read my [mess](https://github.com/Slamy/stm32f4xx-hal/tree/stm32_usb_irq).
As far as I understand, there seem to be two mutual exclusive solutions to IRQs for embedded cortex-m projects. With rtic and without it. I've now seen the example posted here and this seems to be a viable solution. Nonetheless I feel like this doesn't help me and others to fully understand the actual process behind this and being able to understand the internals of Rust in case we stumble upon future problems. The example blinky-timer-irq.rs demonstrates how it can be done as well. But it seems that the USB components are much more difficult to satisfy here. Would it be possible to provide an example based on blinky-timer-irq.rs and usb_serial.rs to show how it is done correctly without relying on rtic?