stm32-rs/stm32f4xx-hal #427: I2S feature breaks debugger connection

kalkyl, 2022-01-31:
I ran into a strange problem on stm32f411 where enabling the "i2s" feature makes my probe run debug connection time out:
```
(HOST) INFO  flashing program (48 pages / 48.00 KiB)
(HOST) INFO  success!
Error: An error with the usage of the probe occured

Caused by:
    Operation timed out
```
This happens even if i don't actually use any the i2s stuff, just enabling the feature in Cargo.toml causes this.
Can anyone reproduce this? Any clues of why this is happening?

burrbull, 2022-01-31:
I think the only one place where it could affect is RCC config.

kalkyl, 2022-01-31:
Yes, but i couldn't find anything featured gated by i2s in there...

YruamaLairba, 2022-03-13:
I have a NucleoF411RE and i don't have the problem. In general, when you use probe connection, there is some things to avoid:
 -  using on of those pins: PA13, PA14, P15, PB3, PB4. They are used for debugging connection.
 -  put the MCU in Sleep mode ( wfi or wfe instruction). probe connection fail if the MCU don't wake up before the timeout.

kalkyl, 2022-03-13:
> I have a NucleoF411RE and i don't have the problem. In general, when you use probe connection, there is some things to avoid:
>
> * using on of those pins: PA13, PA14, P15, PB3, PB4. They are used for [...]

Thanks for checking! Yes i'm well acquainted with working with all this stuff, the thing here is that this problem arises as soon as i just enable the i2s feature in Cargo.toml - without even touching any of the code (which runs fine without).
Haven't tried since the latest release however, so i'll try again and report back whether the problem persists.

dgoodlad, 2022-10-08:
I ran into similar symptoms, and fixed them here https://github.com/samcrow/stm32_i2s/pull/2 . The update to v0.3.0 in master here should include that fix.