stm32-rs/stm32f4xx-hal #455: Add Counter example and/or documentation

techmccat, 2022-02-22:
I've been trying to measure some timings with the counter API but it's not quite as straightforward as I would've hoped.
For example calling Counter.now() consistently returns 0 ticks, which I'm guessing it's because the counter is never enabled, but the docs never mention such requirements.

Because of this the following program (on a F411 blackpill) always returns 0 elapsed time.

```rust
#![no_std]
#![no_main]

use panic_halt as _;

use stm32f4xx_hal as hal;
use cortex_m::asm;
use hal::{prelude::*, timer::Timer};

#[cortex_m_rt::entry]
fn main() -> ! {
    let cp = cortex_m::Peripherals::take().unwrap();
    let sp = hal::pac::Peripherals::take().unwrap();

    let rcc = sp.RCC.constrain();
    let clocks = rcc.cfgr.use_hse(25.MHz()).sysclk(100.MHz()).freeze();
    let counter = Timer::syst(cp.SYST, &clocks).counter_us();

    let start = counter.now();
    for _ in 0..10000000 {
        asm::nop()
    }
    // always returns 0
    let elapsed = counter.now()
        .checked_duration_since(start)
        .unwrap()
        .to_micros();

    loop {}
}
```

burrbull, 2022-02-22:
Thanks for reporting.
Could you try also general-purpose timer? Is problem with SysTick only?

burrbull, 2022-02-22:
Wait. You've created counter, but did not start it.
Add
```rust
counter.start(100.millis()).unwrap()
```
Look at `timer-syst` example.

```rust
#![no_std]
#![no_main]

use panic_rtt_target as _;

use cortex_m_rt::entry;
use rtt_target::{rprintln, rtt_init_print};

use hal::timer::Error;
use stm32f4xx_hal as hal;

use crate::hal::{pac, prelude::*};
use cortex_m::asm;

#[entry]
fn main() -> ! {
    rtt_init_print!();

    let cp = cortex_m::Peripherals::take().unwrap();
    let sp = hal::pac::Peripherals::take().unwrap();

    let rcc = sp.RCC.constrain();
    let clocks = rcc.cfgr.use_hse(25.MHz()).sysclk(48.MHz()).freeze();
    let mut counter = cp.SYST.counter_us(&clocks);
    counter.start(100.millis()).unwrap();

    for _ in 0..10_000_000 {
        asm::nop()
    }

    rprintln!("Elapsed {}", counter.now());

    loop {}
}
```

techmccat, 2022-02-22:
Seems to be working (I don't have a STLink probe on hand so i'm doing LED debugging) as long as sysclk stays under some treshold frequency (looks to be 80MHz by trial and error) beyond which i can get it to work by using the external sysclk clock source.

burrbull, 2022-02-22:
Want to remember that SysTick is only 24bit-wide and does not have prescaler.
So if you want to measure big periods on one timer rotation I'd recommend to use 32-bit general purpose timer.
Just replace `let mut counter = cp.SYST.counter_us(&clocks);` with `let mut counter = cp.TIM2.counter_us(&clocks);`.

closed 2022-02-22 by techmccat