stm32-rs/stm32f4xx-hal #459: Question: is I2sExt trait usable? What it does ? How to use it ?

YruamaLairba, 2022-03-06:
Hi, when searching in the doc if there was a abstraction for duplex i2s or at least for I2Sx_ext, i found [this](url). Unfortunately, i can't figure what is does since there is no documentation about it, not even a single comment in the source. Worst, the intend is very obscure to me because this trait seems is not implemented for any I2S_ext peripheral, and is implemented for any SPI, even those not supporting Full duplex I2S.

burrbull, 2022-03-06:
`xExt` traits is just alternative way to call `new`.
You can create instance with:
```rust
let hal_i2s = I2s::new(dp.SPI3, i2s_pins, &clocks);
// or
let hal_i2s = dp.SPI3.i2s(i2s_pins, &clocks);
```
They are identical

YruamaLairba, 2022-03-07:
Ok, thank you. It's an unfortunate name for something not related to extended I2S.

burrbull, 2022-03-07:
This crate does not contain any i2s implementation. Just wrapper for https://github.com/samcrow/stm32_i2s .
So if you need extended i2s you need add it there first.

YruamaLairba, 2022-03-08:
@burrbull  I think it's may tricky to design a good hal for extended i2s. It's because it share pin/signal with a SPI device. I may even not have sense to make a hal for an i2s_ext alone.

closed 2022-04-25 by YruamaLairba