stm32-rs/stm32f4xx-hal #461: New `gpio::Pin`/`spi::Spi` types appear to kill rust-analyzer

ryankurte, 2022-03-12:
running `rustc 1.61.0`, if you happen to try to alias pins into an SPI device, `rust-analyzer` explodes...

```rust
type Spi1 = Spi<
        SPI1, 
        (
            Pin<'A', 5, Alternate<5> >,
            Pin<'A', 6, Alternate<5> >,
            Pin<'A', 7, Alternate<5> >,
        ),
        TransferModeNormal,
    >;
```

tested with rust-analyzer prerelease via vscode and nighly from rustup. this appears to be a reproduction of https://github.com/rust-analyzer/rust-analyzer/issues/10390, with a suggestion this can be [avoided](https://github.com/rust-analyzer/rust-analyzer/issues/11659#issuecomment-1061773821) by altering the ordering constraint to types before consts, eg. `Pin<MODE = Input<Floating>, const P: char, const N: u8>` instead of `Pin<const P: char, const N: u8, MODE = Input<Floating>>`, though i am not sure whether it is better to change this or wait for a fix...

burrbull, 2022-03-12:
Sadly.
You could use released version of f4xx-hal until RA fixed.

burrbull, 2022-03-12:
@ryankurte 
Is there error if use just:
```rust
type Spi1 =
    Spi<SPI1, (PA5<Alternate<5>>, PA5<Alternate<5>>, PA7<Alternate<5>>), TransferModeNormal>;
```

ryankurte, 2022-03-12:
> You could use released version of f4xx-hal until RA fixed.

unfortunately i'm too far along the bleeding edge to back down at this point 🤣

> Is there error if use just:

brilliant, that resolves it, thanks! also these changes are -super cool-, having a great experience with the updated HAL, hugely appreciate all your effort stm32-rs folks ^_^

burrbull, 2022-03-30:
> @ryankurte Is there error if use just:
> 
> ```rust
> use hal::gpio::{PA5, PA6, PA7};
> 
> type Spi1 =
>     Spi<SPI1, (PA5<Alternate<5>>, PA5<Alternate<5>>, PA7<Alternate<5>>), TransferModeNormal>;
> ```

Now it's possible also
```rust
use hal::gpio::{PA5, PA6, PA7,AF5};
type Spi1 = hal::spi::Spi1<(PA5<AF5>, PA6<AF5>, PA7<AF5>)>;
```

ryankurte, 2022-03-31:
ooh nice! looks like y'all are having some real fun with const types ^_^

thanks for all the help, seems like there's lots of good workarounds

closed 2022-03-31 by ryankurte