stm32-rs/stm32f4xx-hal #480: cannot infer type for type parameter `WORD`

brandonros, 2022-04-13:
```
error[E0282]: type annotations needed for `stm32f4xx_hal::serial::Serial<stm32f4xx_hal::pac::USART1, (stm32f4xx_hal::gpio::Pin<stm32f4xx_hal::gpio::Alternate<stm32f4xx_hal::gpio::PushPull, 7_u8>, 'A', 9_u8>, stm32f4xx_hal::gpio::Pin<stm32f4xx_hal::gpio::Alternate<stm32f4xx_hal::gpio::PushPull, 7_u8>, 'A', 10_u8>), WORD>`
  --> src/main.rs:97:30
   |
97 |         let mut ble_serial = Serial::new(
   |             --------------   ^^^^^^^^^^^ cannot infer type for type parameter `WORD`
   |             |
   |             consider giving `ble_serial` the explicit type `stm32f4xx_hal::serial::Serial<_, (stm32f4xx_hal::gpio::Pin<stm32f4xx_hal::gpio::Alternate<_, 7_u8>, 'A', 9_u8>, stm32f4xx_hal::gpio::Pin<stm32f4xx_hal::gpio::Alternate<_, 7_u8>, 'A', 10_u8>), WORD>`, where the type parameter `WORD` is specified
```

```rust
// Setup BLE UART
        let dp = pac::Peripherals::take().unwrap();
        let tx_pin = gpioa.pa9.into_alternate();
        let rx_pin = gpioa.pa10.into_alternate();
        let mut ble_serial = Serial::new(
            dp.USART1,
            (tx_pin, rx_pin),
            Config::default().baudrate(9600.bps()),
            &clocks,
        ).unwrap();
```

```toml
[dependencies.stm32f4xx-hal]
version = "0.12"
features = ["stm32f407", "rt", "usb_fs", "can"]
```

burrbull, 2022-04-14:
You always can specify appropriate type: `Serial::<_, _, u8>::new(` 
But it strange why it was not done automatically. Maybe you try to call nonexistent method somewhere below.

burrbull, 2024-06-01:
Closing. Reopen if still not resolved.

closed 2024-06-01 by burrbull