stm32-rs/stm32f4xx-hal #482: Compilation Fails on Nightly

TroyNeubauer, 2022-04-15:
I'm trying to use this library as a dependency for an embedded project. Due to limitations in other libraries I have to compile on nightly. 
This works fine for version 0.11

```
git clone https://github.com/stm32-rs/stm32f4xx-hal.git
cd stm32f4xx-hal
git checkout v0.11.0
cargo check --features=stm32f479,usb_fs,sdio-host,can,i2s,fsmc_lcd --examples
```

While on 0.13:
```
git checkout v0.13.0
cargo check --features=stm32f479,usb_fs,sdio-host,can,i2s,fsmc_lcd --examples
```
It fails to compile
```
Compiling stm32f4xx-hal v0.13.0
error[E0432]: unresolved imports `embedded_hal_one::spi::blocking::Operation`, `embedded_hal_one::spi::blocking::Read`, `embedded_hal_one::spi::blocking::Transactional`, `embedded_hal_one::spi::blocking::Transfer`, `embedded_hal_one::spi::blocking::TransferInplace`, `embedded_hal_one::spi::blocking::Write`, `embedded_hal_one::spi::blocking::WriteIter`
  --> /home/troy/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32f4xx-hal-0.13.0/src/spi/hal_1.rs:80:20
   |
80 |         blocking::{Operation, Read, Transactional, Transfer, TransferInplace, Write, WriteIter},
   |                    ^^^^^^^^^  ^^^^  ^^^^^^^^^^^^^  ^^^^^^^^  ^^^^^^^^^^^^^^^  ^^^^^  ^^^^^^^^^ no `WriteIter` in `spi::blocking`
   |                    |          |     |              |         |                |
   |                    |          |     |              |         |                no `Write` in `spi::blocking`
   |                    |          |     |              |         no `TransferInplace` in `spi::blocking`
   |                    |          |     |              no `Transfer` in `spi::blocking`
   |                    |          |     no `Transactional` in `spi::blocking`
   |                    |          no `Read` in `spi::blocking`
   |                    no `Operation` in `spi::blocking`

error[E0433]: failed to resolve: could not find `adc` in `embedded_hal_one`
    --> /home/troy/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32f4xx-hal-0.13.0/src/adc.rs:1078:41
     |
1078 |             impl<PIN> embedded_hal_one::adc::nb::OneShot<pac::$adc_type, u16, PIN> for Adc<pac::$adc_type>
     |                                         ^^^ could not find `adc` in `embedded_hal_one`
...
1101 | adc!(ADC1 => (adc1, ADC_COMMON, 8));
     | ----------------------------------- in this macro invocation
     |
     = note: this error originates in the macro `adc` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0433]: failed to resolve: could not find `adc` in `embedded_hal_one`
    --> /home/troy/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32f4xx-hal-0.13.0/src/adc.rs:1080:92
     |
1080 |                 PIN: embedded_hal::adc::Channel<pac::$adc_type, ID=u8> + embedded_hal_one::adc::nb::Channel<pac::$adc_type, ID=u8>,
     |                                                                                            ^^^ could not find `adc` in `embedded_hal_one`
...
1101 | adc!(ADC1 => (adc1, ADC_COMMON, 8));
     | ----------------------------------- in this macro invocation
     |
     = note: this error originates in the macro `adc` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0433]: failed to resolve: could not find `adc` in `embedded_hal_one`
    --> /home/troy/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32f4xx-hal-0.13.0/src/adc.rs:38:36
     |
38   |               impl embedded_hal_one::adc::nb::Channel<pac::$adc> for $pin {
     |                                      ^^^ could not find `adc` in `embedded_hal_one`
...
1110 | / adc_pins!(
1111 | |     gpio::PA0<Analog> => (ADC1, 0),
1112 | |     gpio::PA1<Analog> => (ADC1, 1),
1113 | |     gpio::PA2<Analog> => (ADC1, 2),
...    |
1123 | |     Vref => (ADC1, 17),
1124 | | );
     | |_- in this macro invocation
     |
     = note: this error originates in the macro `adc_pins` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0433]: failed to resolve: could not find `adc` in `embedded_hal_one`
    --> /home/troy/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32f4xx-hal-0.13.0/src/adc.rs:38:36
     |
38   |               impl embedded_hal_one::adc::nb::Channel<pac::$adc> for $pin {
     |                                      ^^^ could not find `adc` in `embedded_hal_one`
...
1128 | / adc_pins!(
1129 | |     gpio::PC0<Analog> => (ADC1, 10),
1130 | |     gpio::PC1<Analog> => (ADC1, 11),
1131 | |     gpio::PC2<Analog> => (ADC1, 12),
...    |
1134 | |     gpio::PC5<Analog> => (ADC1, 15),
1135 | | );
     | |_- in this macro invocation
     |
     = note: this error originates in the macro `adc_pins` (in Nightly builds, run with -Z macro-backtrace for more info)

Some errors have detailed explanations: E0432, E0433.
For more information about an error, try `rustc --explain E0432`.
error: could not compile `stm32f4xx-hal` due to 5 previous errors
```

I took the features to check with from the command in command in the CI workflow.
The v0.11 features are different because some of them have been added since. 

This is a major issue for me, since I depend on features added since 0.11.

Here is the code for that project if you are interested in building it:
https://github.com/ERFSEDS/novafc-software/tree/131036b44ed766a1066dce81adf6f2566861a6f5/flight

Please let me know what I can do about this

burrbull, 2022-04-15:
It's due to new `embedded-hal` alpha release. My fault. Need to change
```version = "1.0.0-alpha.7"```
 on
```version = "=1.0.0-alpha.7"```

dbrgn, 2022-04-15:
I think this affects both 0.12 and 0.13.

Workaround in downstream projects:

    cargo update -p embedded-hal:1.0.0-alpha.8 --precise 1.0.0-alpha.7

closed 2024-06-01 by burrbull