stm32-rs/stm32f4xx-hal #483: SPI Slave mode

Arnaud078, 2022-04-18:
Hello,

I am working on a project which requires a Raspberry Pi to communicate with a stm32f411. I was thinking of using SPI with the Raspberry as the master. As of now, neither this crate nor the embedded-hal seem to provide this feature. I have forked this repo and implemented a working solution here : https://github.com/Arnaud078/stm32f4xx-hal/commit/4b4baa5ebee1a5b8b0553fbe93ad7343efe0b4d0

I am new to rust, open source and this repo, so I am wondering if that looks good enough for a PR or is there more work to be done ? Is there a cleaner way to achieve SPI slave mode ?

Rgds,
Arnaud

burrbull, 2022-04-19:
If you have ideas with working code you are free to make PR.
It's better to discuss it there.
In first view your code looks good.

burrbull, 2022-04-19:
https://github.com/stm32-rs/stm32f1xx-hal/pull/408

Arnaud078, 2022-04-24:
solved by https://github.com/stm32-rs/stm32f4xx-hal/pull/487

closed 2022-04-24 by Arnaud078